neuropsychology/NeuroKit #3: Neurokit2: Call for help

DominiqueMakowski, 2019-10-30:
The original [neurokit](https://github.com/neuropsychology/NeuroKit.py) package started as a small project as I was learning python, but ended having far more success than I originally expected, far beyond its quality of implementation.

Thus, I think it's better to restart the project from scratch, in a more open and collaborative manner, to create the **top toolbox for neurophysiological data processing**. If you're interested by such project, whether it is because you need it, because you want to learn python, because you need published papers or technical achievements to add to your CV, you're very welcome to hop aboard! Now is the right time, just let us know with a 👍 react or a response to this thread, mentioning features would you like to see, or how could you eventually help! 

@hcp4715 @sangfrois

sangfrois, 2019-10-30:
You can count on me for all the reasons above, Dominique!

hcp4715, 2019-10-30:
I am really happy to join in. but I am still pretty new to Python ;-).

hosseinhamidi, 2019-10-31:
I'll also be happy if you need any help as I just started to work on ECG signals with Python.

GanshengT, 2019-11-25:
I am currently working on a corticomuscular coherence (CMC). I am interested in further development of NeuroKit. NeuroKit might serve as an analysis platform in my project. I am glad to develop.

hosseinhamidi, 2019-11-25:
Hi,

I am interested to help you in developing this toolbox as kind of these
methods are related to my project to process my ECG and PPG signals. Please
let me know how we can shape a cooperation.

Regards,
Hossein

Tam-Pham, 2019-11-25:
Count me in! I'm interested particularly in the HRV. Let us know how we can help!

DominiqueMakowski, 2019-11-25:
Great! I am currently trying to have a working a stable and top-standards setup for the package and the github repo, i.e., that passes tests, that is on pypi, that has automated documentation etc. Once that is done, we can start thinking about the structure and organisation! I'll keep you guys updated ☺️ 

In the meantime, if you are already using processing your signals, 1) be on the lookout for the existing packages/programs that could help us in terms of features or implementation and 2) if you are writing some code, try to make it as modularized as possible (i.e., creating functions) so we could reuse them here.

💪

DominiqueMakowski, 2019-12-06:
Small update 🎉  I think the repo is now starting to be in shape. Testing, automated code reviews and documentation generation seem to be working. I will now start porting some of the useful functions from NeuroKit1 to here. We already ported (and strongly improved) some of them, such as `read_acqknowledge()` and `events_find()`. Will no go ahead with the biophysio stuff. 

However, before doing that, I'd like to test the contributing-friendliness of the repo. I put together a [document]() with guidelines for contributing (the source is [here](https://github.com/neuropsychology/NeuroKit/blob/master/.github/CONTRIBUTING.rst)). Could you guys give it a look, see if it's understandable, useful and all (and improve it if need be!). 

Following this workflow, could you then all **add your name** to the [contributors section](https://github.com/neuropsychology/NeuroKit/blob/master/AUTHORS.rst) and **make a PR** to test the workflow of PRing to dev, and from dev to master? Thanks a lot! Let me know if there are any issues ☺️ 

@sangfrois @hcp4715 @GanshengT @hosseinhamidi @PhilippeBlonde @zen-juen

DominiqueMakowski, 2019-12-06:
currently the author's page is separated into 2 sections (core team and contributors). The latter refers to people helping and contributing to the package (and you, since you've motivated me to start this) while the former refers to people that are sort of responsible for the package (maintenance, addressing issues, bugfixes etc.), which requires a good understanding of the package as a whole (tests, setup, etc.). For now, I'd suggest you add yourself as contributors, and you'll see in time how you get familiar with the package ☺️

GanshengT, 2019-12-06:
You might also want to indicate the workflow to synchronize the fork by merging the upstream repository into the folk.

DominiqueMakowski, 2019-12-06:
> You might also want to indicate the workflow to synchronize the fork by merging the upstream repository into the folk.

Do you have any suggestions on how to clarify that?

sangfrois, 2019-12-06:
Hi Dom,

Thanks for reaching out again, I've read your tutorial for contributing and I think the intention is aligned with the effect it has, i.e. I felt compelled to contribute and welcomed in the process. I hope my pull request has worked alright. I forked the repo. Then, I've added my name to the contributors and commited this change on a local dev branch. Afterwards, I've merged it with the master. I then pushed it on the remote 
`git remote -v show
dev     https://github.com/sangfrois/NeuroKit (fetch)
dev     https://github.com/sangfrois/NeuroKit (push)
dev     https://github.com/neuropsychology/NeuroKit (push)
origin  https://github.com/sangfrois/NeuroKit.git (fetch)
origin  https://github.com/sangfrois/NeuroKit.git (push)`

I've been busy trying to get my grades up lately, but still

DominiqueMakowski, 2019-12-07:
@sangfrois well to be honest my knowledge of git command line is super limited (GitHub desktop FTW 😁), so I am not sure. However, I don't see your PR on this repo, indicating that something is missing.

The steps are roughly 1) fork, 2) switch to dev branch 3) make changes 4) push 5) make PR (so that the changes of your fork are pushed to the dev branch here)

If anyone knows the corresponding git commands please feel free to clarify!

sangfrois, 2019-12-07:
Oh right !! I see there's no corresponding git commands ; I think I'm only missing step 5 which has to be done on GitHub directly! Thanks for the clarification, I'll make a PR first thing when I wake up!

closed 2020-03-12 by DominiqueMakowski

kassyray, 2020-06-11:
Hi @DominiqueMakowski - I'm a little late on seeing this post but I still see lots of activity on this project. I'm an MSc student working on signal processing (predominantly EDA), but would love to get involved!

DominiqueMakowski, 2020-06-12:
Hi @kassyray no worries it's never too late, happy to have you onboard ☺️ Even though already *big* in terms of feature, I consider this project still in its very early days with a long way to go. You hop on the train as it is barely leaving the station 🚄 

> predominantly EDA

Fantastic, especially as this is an area that has a lot of potential improvements. 

Do not hesitate to dig into the repo participate to discussions ask questions suggest ideas etc., and on our side, we'll try to ping you whenever possible

commit b687a4fa (2020-03-10): Fix #3