neuropsychology/NeuroKit #5: Example data

DominiqueMakowski, 2019-10-31:
I'd say that a start will be to 1) implement some data reading function(s) [can start with the `read_acqknowledge` in NeuroKit1] and 2) upload some test data, which we will be able to use as a testcase, and also for documentation and tutorials.

For 1), we could also think about a wrapper around the [wfdb](https://github.com/MIT-LCP/wfdb-python/tree/master/wfdb/io) reader to do tests on data from https://physionet.org/

For 2) I will probably be able to record and upload some data using Biopac and the [empathica](https://support.empatica.com/hc/en-us/categories/200023126-E4-wristband) wristband, for a simple paradigm (e.g., neutral vs. emotional pictures), in the next few weeks. 

@hcp4715 @sangfrois may I know if you have any particular recording device that you used/use/might use?

JohnDoenut, 2019-10-31:
Hi @DominiqueMakowski,

regarding 1), I started implementing a [wrapper](https://github.com/JohnDoenut/biopeaks/blob/master/backend_test.py) around wfdb for testing purposes a while back.

regarding 2), we're using the [Bitalino](https://bitalino.com/en/) at our lab and I could provide some sample data for ECG and breathing (although that might not be necessary if we grab data from wfdb).

sangfrois, 2019-11-01:
Hey @DominiqueMakowski ; of course, i can tell you : my lab collects signals with various amplifiers. We have a biopac set and record ECG, Resp, EDA, pulse. Stationned at [UNF](https://unf-montreal.ca/infrastructure/) we acquire them simultaneously with fMRI and even M/EEG (so a lot of options).  The team will soon openly release data from a small sample (n<15) scanned intensively and also publish tools of analyses via our github. 

* So yeah, i can surely give a look at `read_acknowledge` in neurokit1 .

hcp4715, 2019-11-01:
Hi, @DominiqueMakowski ,

Our lab only uses BioPac to record EDA and ECG data, but we usually give electric shocks.

DominiqueMakowski, 2019-12-16:
data can be now stored in the data folder, + we start to have some nice simulators. 

Once we start on adding more processing functions, we will test them using different data sources, I'll let you know ☺️

closed 2019-12-16 by DominiqueMakowski