neuropsychology/NeuroKit #13: can we use this library to do the respiratory rate estimation via the ECG data?

mousehaohao, 2019-12-05:
Hi Team,
    Great to find this library that can process biosignals, seems it still under construct, can we use this library to do the respiratory rate estimation via the ECG data?

DominiqueMakowski, 2019-12-05:
Hi @mousehaohao, I *think* you are referring to the rate of the slower frequency bands of the heart rate variability (that are correlated with respiration)? 

If that's the case, then in the (hopefully near) future, yes, neurokit will be able to do that. I'd suggest you check [Neurokit 1](https://github.com/neuropsychology/NeuroKit.py), to have an idea of what we plan to do (although the goal is to go higher and beyond in terms of features and quality).

Do stay tuned here, and do not hesitate to share your input the goal is to have an open project ☺️

mousehaohao, 2019-12-06:
Sure, thanks for the quick response, I'll check the Neurokit1 first and hope I can join this project someday :)

closed 2019-12-06 by mousehaohao

DominiqueMakowski, 2019-12-06:
@mousehaohao if you are interested in contributing and helping this project, feel free to comment on #3 😉