neuropsychology/NeuroKit #15: Ideas / Ressources for Respiration

DominiqueMakowski, 2019-12-06:
### Features

- [ ] RSP rate
- [ ] ...

### Papers

- [Noto et al., 2018](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC6150778/)

GanshengT, 2019-12-06:
One feature that arouses great research interest is the capacity of plotting EMG-EEG topograph. For example, the EMG-EEG connectivity analysis

DominiqueMakowski, 2019-12-06:
You mean MEG/EEG I think? Doesn't [mne](https://mne.tools/stable/generated/mne.viz.plot_topomap.html) do that? And how does it relate to respiration (the topic of this issue) 😁 ?

GanshengT, 2019-12-06:
It’s indeed Electromyography (EMG) - EEG connectivity analysis. In the case of respiration, information of respiratory muscle and EEG are recorded simultaneously. Their correlation could possibly reveal respirator mechanism😊. MNE is not oriented to do such analysis.

DominiqueMakowski, 2019-12-06:
Mmh that's interesting, never heard of that! Are you working on this?

GanshengT, 2019-12-06:
To be specific, I currently worked on the connectivity between EMG and EEG during hand gripping paradigm, but I equally covered this field.

DominiqueMakowski, 2019-12-06:
That's great! Looking forward to have your input ☺️

JohnDoenut, 2019-12-06:
Is the idea to extend/improve the existing [breathing functions](https://github.com/neuropsychology/NeuroKit.py/blob/master/neurokit/bio/bio_rsp.py) or to start from sratch?

DominiqueMakowski, 2019-12-06:
@JohnDoenut Well there hasn't been much done on that side in NK1. We can use that as a basis, but I believe there is room for improvement!

JohnDoenut, 2019-12-08:
For the estimation of breathing rate and breathing amplitude, I'm happy to contribute an [algorithm](https://github.com/JohnDoenut/biopeaks/blob/master/biopeaks/resp.py) based on this [paper](https://www.ncbi.nlm.nih.gov/pubmed/30074906). It does what has previously been done by Nk1's rsp_process(): it identifies inhalation peaks and exhalation troughs which are then used to calculate statistics (summary, instantaneous) for the breathing signal. Trying to send a PR asap.

DominiqueMakowski, 2019-12-16:
closing this as thanks to the awesome contribution of @JohnDoenut we have  a strong basis for rsp ☺️

closed 2019-12-16 by DominiqueMakowski