neuropsychology/NeuroKit #20: Simulate synthetic physiological signals (ECG, RSP, EDA/SCR)

DominiqueMakowski, 2019-12-06:
I want to add functions to simulate synthetic bio signals (e.g., ECG, RSP and EDA), which would be convenient for documentation, tests and all. I found some useful materials and implementations, but there are all in matlab... I'd like to translate it to python, but I don't know matlab enough... Anybody knows 😅? Or knows someone who does to give it a look?

## ECG

There is a [new algorithm](https://physionet.org/content/ecgsyn/1.0.0/) implemented in different languages, including [matlab](https://physionet.org/content/ecgsyn/1.0.0/Matlab/ecgsyn.m). The code doesn't look too complicated to translate, but I don't know where to start (@suklamaa)

## RSP

For respiration, I found this 
 https://github.com/zelanolab/breathmetrics/blob/master/simulateRespiratoryData.m matlab implementation.

## EDA 

I found [Bach's (2010)](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC2877881/) paper which attempts to model the canonical SCR.  The mathematical definition is in Appendix A, and some of the pieces are apparently implemented in matlab ([here](https://sourceforge.net/p/scralyze/code/HEAD/tree/branches/version_b2.1.8/scr_bf_crf.m) and [here](https://sourceforge.net/p/scralyze/code/HEAD/tree/branches/version_b2.1.8/scr_bf_infbs.m))

DominiqueMakowski, 2019-12-06:
I am unsure about one line, does anyone know how to translate the last matlab line into python?

(assuming that dw1 and dw2 are vectors and c1 and c2 two floats):

```matlab
Nrr = 64
Hw1 = sig1*exp(-0.5*(dw1/c1).^2)/sqrt(2*pi*c1^2);
Hw2 = sig2*exp(-0.5*(dw2/c2).^2)/sqrt(2*pi*c2^2);
Hw = Hw1 + Hw2;

Hw0 = [Hw(1:n/2); Hw(n/2:-1:1)];
```

```python
Nrr = 64

Hw1 = sig1*np.exp(-0.5*(dw1/c1)**2)/np.sqrt(2*np.pi*c1**2)
Hw2 = sig2*np.exp(-0.5*(dw2/c2)**2)/np.sqrt(2*np.pi*c2**2)
Hw = Hw1 + Hw2

Hw0 = np.array([Hw*np.arange(1, Nrr/2), Hw*np.arange(Nrr/2, -1, 1)]) ???
```

suklamaa, 2019-12-06:
Dear dom
Please change this way in python. np.arange(Nrr/2,1,-1)

DominiqueMakowski, 2019-12-07:
Thanks! but there something wrong somewhere, I get a broadcasting together as the `np.arange` array is of different size from the `Hw` array. Here's the reproducible matlab code and the python code that I have so far:

```m
# Here are the default values for the arguments
flo=0.1;
fhi=0.25;
flostd=0.01;
fhistd=0.01;
lfhfratio=0.5;
hrmean=60;
hrstd=1;
sfrr=1;
n=64;
# ----------

w1 = 2*pi*flo;
w2 = 2*pi*fhi;
c1 = 2*pi*flostd;
c2 = 2*pi*fhistd;
sig2 = 1;
sig1 = lfhfratio;
rrmean = 60/hrmean;
rrstd = 60*hrstd/(hrmean*hrmean);

df = sfrr/n;
w = [0:n-1]'*2*pi*df;
dw1 = w-w1;
dw2 = w-w2;

Hw1 = sig1*exp(-0.5*(dw1/c1).^2)/sqrt(2*pi*c1^2);
Hw2 = sig2*exp(-0.5*(dw2/c2).^2)/sqrt(2*pi*c2^2);
Hw = Hw1 + Hw2;
Hw0 = [Hw(1:n/2); Hw(n/2:-1:1)];
```

```python
# ----------
# Here are the default values for the arguments
flo=0.1
fhi=0.25
flostd=0.01
fhistd=0.01
lfhfratio=0.5
hrmean=60
hrstd=1
sfrr=1
n=64
# ----------

w1 = 2*np.pi*flo
w2 = 2*np.pi*fhi
c1 = 2*np.pi*flostd
c2 = 2*np.pi*fhistd
sig2 = 1
sig1 = lfhfratio
rrmean = 60/hrmean
rrstd = 60*hrstd/(hrmean*hrmean)

df = sfrr/n
w = np.arange(n-1)*2*np.pi*df
dw1 = w-w1
dw2 = w-w2

Hw1 = sig1*np.exp(-0.5*(dw1/c1)**2)/np.sqrt(2*np.pi*c1**2)
Hw2 = sig2*np.exp(-0.5*(dw2/c2)**2)/np.sqrt(2*np.pi*c2**2)
Hw = Hw1 + Hw2

Hw0 = [Hw*np.arange(1, n/2), Hw*np.arange(n/2, 1 ,-1)]
```

When I run the above, I have Hw that is of size 63, but the range of the generated vector is only  31. 

Does `[Hw(1:n/2); Hw(n/2:-1:1)]` mean a multiplication of `Hw` with an array?

hungpham2511, 2019-12-07:
Hi Dom, it basically is a vertical stack (vstack) in numpy. See here
```
>> Hw = [1:1:63]
Hw =

 Columns 1 through 24:

    1    2    3    4    5    6    7    8    9   10   11   12   13   14   15   16   17   18   19   20   21   22   23   24

 Columns 25 through 48:

   25   26   27   28   29   30   31   32   33   34   35   36   37   38   39   40   41   42   43   44   45   46   47   48

 Columns 49 through 63:

   49   50   51   52   53   54   55   56   57   58   59   60   61   62   63

>> n = 64
n =  64
>> [Hw(1:n/2); Hw(n/2:-1:1)]
ans =

 Columns 1 through 24:

    1    2    3    4    5    6    7    8    9   10   11   12   13   14   15   16   17   18   19   20   21   22   23   24
   32   31   30   29   28   27   26   25   24   23   22   21   20   19   18   17   16   15   14   13   12   11   10    9

 Columns 25 through 32:

   25   26   27   28   29   30   31   32
    8    7    6    5    4    3    2    1
```

So in Python should be `np.vstack((Hw[:n/2], Hw[n/2-1::-1]))`.

JohnDoenut, 2019-12-07:
Matlab

```
>> Hw = 1:10

1     2     3     4     5     6     7     8     9    10

>> Hw(1:5)

1     2     3     4     5

>> Hw(5:-1:1)

5     4     3     2     1

>> Hw0 = [x(1:5); x(5:-1:1)]

     1     2     3     4     5
     5     4     3     2     1
```

Python
```
>> Hw = np.arange(1, 11)

array([ 1,  2,  3,  4,  5,  6,  7,  8,  9, 10])

>> Hw[0:5]

array([1, 2, 3, 4, 5])

>> Hw[4::-1]

array([5, 4, 3, 2, 1])

>> Hw0 = np.row_stack((x[0:5], x[4::-1]))
 
array([[1, 2, 3, 4, 5],
       [5, 4, 3, 2, 1]])
```

DominiqueMakowski, 2019-12-07:
So, matlab's `Hw0 = [Hw(1:n/2); Hw(n/2:-1:1)];` should be equivalent to `Hw0 =  np.row_stack((Hw[0:int(n/2)], Hw[int(n/2)-1::-1]))`  (taking into account the 1-indexing in matlab)? Which returns a 2D array.

The line after says: `ph0 = 2*pi*rand(n/2-1,1);` Is this equivalent to `h0 = 2*np.pi*np.random.uniform(n/2-1, 1)` (it returns a single float)?

DominiqueMakowski, 2019-12-07:
> The line after says: ph0 = 2*pi*rand(n/2-1,1); Is this equivalent to h0 = 2*np.pi*np.random.uniform(n/2-1, 1) (it returns a single float)?

That doesn't seem right, as in the line after, matlab `ph = [ 0; ph0; 0; -flipud(ph0) ]; ` the what-I-believe-to-be pytnon's equivalent np.flipud() requires an array > 1d 😕

JohnDoenut, 2019-12-07:
I would just run all this in Matlab and look at each variable to get an idea of what's going on. Note that if you don't have (or want :grin:) a Matlab licencse, [Octave](https://www.gnu.org/software/octave/) is a free alternative that has mostly the same syntax. Sorry for the overly generic response :)

DominiqueMakowski, 2019-12-07:
Good idea!

(Never thought I'd have to learn matlab by working on a python project 😠😅)

DominiqueMakowski, 2019-12-07:
I gave it my best shot and managed to obtain a code that *runs* and, until almost the end, seemed to resemble the octave/matlab results. However, I am not sure what it does at the end (I suppose it calls the small `derivsecgsyn` from the second file but I don't understand what that function returns), and the final result... doesn't really look like an ECG singal 😞 (see the plot)

https://github.com/neuropsychology/NeuroKit/blob/dev/scripts/ecgsynth.ipynb

hcp4715, 2019-12-07:
Hi, Dom, 

> I found Bach's (2010) paper which attempts to model the canonical SCR. The mathematical definition is in Appendix A, and some of the pieces are apparently implemented in matlab (here and here)

As you mention Bach's work, their latest PsPM package (in matlab) is here:
https://github.com/bachlab/PsPM

To my knowledge, Bach's had compared his model with LEDALab and his model: https://www.sciencedirect.com/science/article/pii/S0301051114001847. In NeuroKit 1, the algorithm used was based cvxEDA, an improved version of LEDAlab's continuous deconvolution analysis
(CDA).  Not sure we plan to include Bach's canonical model.

DominiqueMakowski, 2019-12-08:
For the analysis of EDA, I definitely prefer having the most recent and accurate technique 💪  Here, the goal was rather to eventually add a function to generate a simulated version of biosignals. By implementing the canonical SCR, we could have then think of a function that would generate a signal of a given duration with a given number of SCRs.

The goal of having signal generation functions is to be able to easily demonstrate, showcase the processing functions without having to download and use a test dataset, and also to be able to test the processing functions in a reproducible and controlled environment (for example, if we had a proper ECG simulator, we could test all the processing functions under different controlled scenarios of noise, rate, anomalies etc.). I am not sure if something has be attempted for PPG and EMG

DominiqueMakowski, 2019-12-16:
Closing this as we almost have it all (still missing #39 and #25)

closed 2019-12-16 by DominiqueMakowski