neuropsychology/NeuroKit #25: Simulating PPG signal (implementing an equation)

DominiqueMakowski, 2019-12-08:
Let's translate this to python:

[image]

DominiqueMakowski, 2019-12-08:
Here's a sum of two gaussians. But I don't understand the parameters that they suggest in the paper 🤔 

```python
    index = np.linspace(-9, 9, num=100)
    normal1 = scipy.stats.norm.pdf(index, loc=-2, scale=2)/1.5
    normal2 = scipy.stats.norm.pdf(index, loc=2, scale=1.5)
    ppg = normal1+normal2
    pd.Series(ppg).plot()
```

[image]

JanCBrammer, 2020-03-20:
@DominiqueMakowski, from which paper is the snippet you posted above? What they do is basically just making the sum of two Gaussians (yk) look like a recorded PPG signal (PPGk) with brute force optimization. That is, they tweak the parameters of the Gaussians until there is a minimal difference between the recorded and simulated PPG waveform. Conceptually the same as linear regression: tweak the intercept and slope parameters until the residuals are minimized. I guess for our simulator we would want something more generic that doesn not depend on recorded data?

DominiqueMakowski, 2020-03-21:
>  from which paper is the snippet you posted above? 

Paper is http://150.162.46.34:8080/icassp2015/pdfs/0000967.pdf

> I guess for our simulator we would want something more generic that does not depend on recorded data?

Yeah, although as a start we could 1) have a function to construct one PPG "beat" (with different parameters for the amplitude etc.) and 2) the simulator would then make it periodic (similarly to what the Daubechies method do for ecg_simulate, it just takes the wavelet and repeats it to achieve a given heart_rate). Until someone comes with a proper sophisticated algorithm to generate a realistic signal.

JanCBrammer, 2020-03-21:
Thanks for the paper! The following repository seems like a very useful resource as well, linkting it here for future reference: https://github.com/peterhcharlton/pwdb/tree/master/pwdb_v1.0

DominiqueMakowski, 2020-03-21:
Thanks for that link! Wasn't aware of that toolbox

It seems that what we need is [mentioned here](https://peterhcharlton.github.io/pwdb/algorithms.html) as `AorticFlowWave`: "AorticFlowWave is a tool for creating synthetic aortic flow waves with different properties, such as blood pressure and photoplethysmogram pulse waves"

The *matlab* code is [here](https://raw.githubusercontent.com/peterhcharlton/pwdb/master/pwdb_v0.1/Additional%20Functions/AorticFlowWave.m). Tagging the de facto translation expert @Tam-Pham 😁

JanCBrammer, 2020-03-24:
These papers describe how to model PPG without many physiological assumptions (unlike the `AorticFlowWave`). No code though, only equations (maybe some brave soul has some time on their hands to take them on) 😬

https://www.frontiersin.org/articles/10.3389/fphys.2017.00701/full

https://arxiv.org/abs/1605.01805

DominiqueMakowski, 2020-03-24:
That seems like a job for superman

JanCBrammer, 2020-03-24:
Started hacking on some "naive approximation" of PPG: The PPG wave is described with four landmarks: wave onset, location of the systolic peak, location of the dicrotic notch and location of the diastolic peak. These landmarks are defined based on their x and y coordinates (in a time series). These coordinates are then interpolated at the desired sampling rate to obtain the PPG signal.

```python
import matplotlib.pyplot as plt
import numpy as np
from scipy.interpolate import interp1d

sfreq = 100
heartrate = 70    # in bpm
duration = 120    # in seconds

# at the requested sampling rate, how long is a period at the requested heart
# rate and how often does that period fit into the requested duration?
period = 60 / heartrate   # in seconds
n_period = int(np.floor(duration / period))
periods = np.ones(n_period) * period
# seconds at which periods end
x_onset = np.cumsum(periods) - period
y_onset = 0

x_sys = x_onset + 0.3 * period
y_sys = y_onset + .2

x_notch = x_onset + 0.4 * period
y_notch = y_onset + .05

x_dia = x_onset + 0.45 * period
y_dia = y_onset + .04

x_all = np.concatenate((x_onset, x_sys, x_notch, x_dia))
x_all.sort(kind="mergesort")
x_all = np.rint(x_all * sfreq).astype(int)
y_all = [y_onset, y_sys, y_notch, y_dia] * n_period

f = interp1d(x_all, y_all, kind="cubic", bounds_error=False,
              fill_value=([y_all[0]], [y_all[-1]]))
samples = np.arange(0, duration * sfreq)
ppg = f(samples)

fig, ax = plt.subplots(nrows=2, ncols=1, sharex=True)

ax.scatter(x_all, y_all, c="r")
ax.plot(ppg)

```
The result looks ok, imagine adding some frequency modulation (sinusoidally modulated spacing of systolic peaks), adding noise, baseline fluctiation etc.

[image]

DominiqueMakowski, 2020-03-24:
nice! 🎉 And interesting usage of cubic interpolation! I wonder it's computationally different from let's say using a mixture of two gaussians as they suggest in the first paper 🤔 but let's first add this one, I'll let you make a PR ☺️

JanCBrammer, 2020-03-24:
> I wonder it's computationally different from let's say using a mixture of two gaussians as they suggest in the first paper

We could also try to fit a gaussian process to the red dots. There should be a kernel that does a nice job approximating the PPG: https://blog.dominodatalab.com/fitting-gaussian-process-models-python/

JanCBrammer, 2020-03-26:
Stuff starts looking pretty solid. Added respiratory sinus arrythmia (frequency modulation), baseline drift, motion artifacts, powerline interference and short bursts of high frequency noise. `signal_distort()` and `signal_simulate()` where really helpful btw (made some modifications)! Have to clean up a few things before making the PR. Should be this weekend.

These are 40 seconds of PPG at sampling rate of 1000 and heart rate of 70:

[image]

DominiqueMakowski, 2020-03-27:
Man that looks really impressive! 😍

DominiqueMakowski, 2020-03-29:
I'll close that as we have a first version, we can open new issues if new ideas/bugs come out ☺️

closed 2020-03-29 by DominiqueMakowski