neuropsychology/NeuroKit #31: Resources: ECG processing and features

DominiqueMakowski, 2019-12-11:
Collection of resources to help us getting the ECG aspect right, feel free to add more:

-----

## HRV

- [Guidelines for Reporting Articles on Psychiatry and Heart rate variability](https://www.nature.com/articles/tp201673)

JohnDoenut, 2019-12-30:
To get us started with ECG processing I'm happy to add an [ECG peak detection](https://github.com/JohnDoenut/biopeaks/blob/master/biopeaks/ecg.py) algorithm to the `ecg` module. I recently benchmarked the algorithm on the [Glasgow University Database (GUDB)](http://researchdata.gla.ac.uk/716/) (see [5. Changelog](https://github.com/JohnDoenut/biopeaks) for details). Then the users could already get heart rate and -period out of their ECG data and we would have a basis for building HRV analyses. If this sounds like a good idea I'll get working on a PR soonish in the new year :) 

On a related note, here's a nice [overview](https://www.biorxiv.org/content/10.1101/722397v3.full) of popular ECG peak detection algorithms and their performance on established benchmarks.

DominiqueMakowski, 2019-12-31:
@JohnDoenut That's great 😍 Looking forward to your PR in the next decade 😁

DominiqueMakowski, 2019-12-31:
@Tam-Pham It would be useful to get an overview of the existing pipelines for ECG cleaning (parameters of filters etc..). For instance, biosspy uses only simple filtering:

https://github.com/PIA-Group/BioSPPy/blob/e65da30f6379852ecb98f8e2e0c9b4b5175416c3/biosppy/signals/ecg.py#L69-L76

But there are [more](https://github.com/neuropsychology/NeuroKit#general) packages and surely some publications about it :wink:

JohnDoenut, 2020-01-03:
See #65.

DominiqueMakowski, 2020-02-26:
Closing this in favour of the more specialized issues.

closed 2020-02-26 by DominiqueMakowski