neuropsychology/NeuroKit #39: ECG simulation with ecgsyn

DominiqueMakowski, 2019-12-16:
- [ecgsyn](https://physionet.org/content/ecgsyn/1.0.0/): Better simulation of the QRS complex: there is a new algorithm implemented in different languages, including matlab [code here](https://physionet.org/content/ecgsyn/1.0.0/Matlab/ecgsyn.m).

I started translating the code [here](https://github.com/neuropsychology/NeuroKit/blob/dev/neurokit2/ecg/ecg_ecgsynth.py), but the results are not what it should like. IMO something gets round around [here](https://github.com/neuropsychology/NeuroKit/blob/dev/neurokit2/ecg/ecg_ecgsynth.py#L79).

Tam-Pham, 2019-12-20:
@duylp This is the [python file ](https://github.com/neuropsychology/NeuroKit/blob/dev/scripts/WIP_ecg_ecgsynth.py) and the 2 matlab codes I showed you yesterday are [ecgsyn](https://physionet.org/content/ecgsyn/1.0.0/Matlab/ecgsyn.m) and [deriveecgsyn](https://physionet.org/content/ecgsyn/1.0.0/Matlab/derivsecgsyn.m). 

ya my best bro!

@DominiqueMakowski, @duylp has been helping me to run the Matlab code and he's an awesome guy who now will continue to help with translating the `ode45` function =)))))

DominiqueMakowski, 2019-12-20:
@duylp ma man 👐 🙌

duylp, 2019-12-23:
@DominiqueMakowski @Tam-Pham I'm happy to help.

Can you explain what [this line](https://github.com/neuropsychology/NeuroKit/blob/4537c744e9d6ee07ba06ba363d1fb90329ec316b/neurokit2/ecg/ecg_ecgsynth.py#L116) does in your code? It looks weird to me. I also got an out of bound error when indexing using the original code.

The translation is done. I got the output signal below from my code. You have any idea how it should look like? Never seen an ECG before 😅
[image]

DominiqueMakowski, 2019-12-24:
> Can you explain what this line does in your code? It looks weird to me. I also got an out of bound error when indexing using the original code.

I got an error when using the original code too, thus I tried to tweak it not to have the error... But not sure if that made sense, as I don't know the purpose of this line...

> The translation is done. I got the output signal below from my code. You have any idea how it should look like? Never seen an ECG before 😅

Bravo! It's not *far* from an ECG signal, in that that is seems like the components are there (the peak, corresponding to the R peak, and the P and S waves (before and after the peak). However, there seem to be 2 issues. 1) there is this big negative low-freq negative deflection that shouldn't be there (a typical ECG is usually fairly flat, i.e., doesn't drift up or down) and 2) the Q, R and S components (the main "peaks"/waves) are also king of misaligned. It seems like all the cardiac complexes are rotated 😕 I wonder why

But thanks a lot for giving it a look ☺️ 

[image]

[image]

DominiqueMakowski, 2019-12-24:
In the [software description](https://physionet.org/content/ecgsyn/1.0.0/Matlab/#files-panel), they say that they have a Java implementation with [an app](https://archive.physionet.org/physiotools/ecgsyn/Java/ecgsyn-java.html), in which they have a sample image of what it should look like (I suppose):

[image] 

Maybe we could check in the C or java code what's up with that line that errors to see how

duylp, 2019-12-25:
This is the output from their Matlab code. There are still some parts I haven't been able to translate correctly to Python (like the line of code I mentioned above) which may cause the differences. I will update if there is any news
[image]

DominiqueMakowski, 2019-12-26:
The matlab output looks correct! You think the error's origin is here?

https://github.com/neuropsychology/NeuroKit/blob/4537c744e9d6ee07ba06ba363d1fb90329ec316b/neurokit2/ecg/ecg_ecgsynth.py#L115-L116

duylp, 2019-12-26:
In my code I use `w0 = 2*np.pi/rrn[min(ip, len(rrn)-1)]` to avoid the out of bound error. The original code should work fine and there is something wrong with `rrn`.
The difference between the python and matlab starts from here: https://github.com/neuropsychology/NeuroKit/blob/4537c744e9d6ee07ba06ba363d1fb90329ec316b/neurokit2/ecg/ecg_ecgsynth.py#L61-L74

DominiqueMakowski, 2019-12-27:
Maybe things get wrong at `rr = signal_resample(rr0, sampling_rate=1, desired_sampling_rate=sfint)`. In the original matlab code it's `rr = interp(rr0,sfint);`. I can't compare the two, but maybe it would be better to use [`signal_interpolate()`](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/signal/signal_interpolate.py)? Altough in theory signal_resample should work 🤔

duylp, 2019-12-27:
[Matlab interp](https://www.mathworks.com/help/signal/ref/interp.html#d117e92068) uses a complex algorithm (applying a low pass filter) for upsampling. As far as I know, there are no existing one-line functions in numpy or scipy that are similar to it. I don't mind reimplementing it in Python but also wonder why the implementation is so sensitive to different methods of upsampling?

hungpham2511, 2019-12-27:
@duylp Naive upsampling can introduce numerical artifacts because the possible frequency spectrum is suddenly enlarged (due to the faster sampling rate). Because of that, it is common to run a low-pass filter through the up-sampled signal (at possible half the Nyquist freq of the original signal). Still, this approach is rather too straight-forward and there should be a better way.

hungpham2511, 2019-12-27:
Hi guys, so the culprit is the difference between Matlab `rem` and `numpy.remainder`, I just tried this this:
```
-    dti = np.remainder(ta - ti, 2*np.pi)
+    dti = (ta - ti) - np.round((ta - ti) / 2 / np.pi) * 2 * np.pi
```
and got this:
[image]

Tam-Pham, 2019-12-27:
@duylp @hungpham2511  THANKS GUYS!!! Ya the best <3

DominiqueMakowski, 2019-12-27:
🎉 🎉 Wow, congrats!!! That was no easy feat! I, and the future users, are super grateful to you guys ☺️ 

@duylp please add yourself to the [package contributors](https://github.com/neuropsychology/NeuroKit/blob/master/AUTHORS.rst), and stay tuned for more challenges to come 😁 😁

DominiqueMakowski, 2019-12-30:
Closing this as this is now officially in ☺️ great work!

closed 2019-12-30 by DominiqueMakowski