neuropsychology/NeuroKit #40: Resources: complexity and entropy measures

DominiqueMakowski, 2019-12-16:
## Measures

### Entropy

- [ ] Shannon entropy ([pyEntrp](https://github.com/nikdon/pyEntropy))
- [ ] Sample entropy ([nolds](https://github.com/CSchoel/nolds), [pyEntrp](https://github.com/nikdon/pyEntropy), [EntroPy](https://github.com/raphaelvallat/entropy))
- [ ] multiscale entropy (MSE) ([pyEntrp](https://github.com/nikdon/pyEntropy))
- [ ] Composite multiscale entropy ([pyEntrp](https://github.com/nikdon/pyEntropy))
- [ ] Permutation entropy ([pyEntrp](https://github.com/nikdon/pyEntropy), [EntroPy](https://github.com/raphaelvallat/entropy))
- [ ] Multiscale permutation entropy ([pyEntrp](https://github.com/nikdon/pyEntropy))
- [ ] Spectral entropy ([EntroPy](https://github.com/raphaelvallat/entropy))
- [ ] Singular value decomposition entropy ([EntroPy](https://github.com/raphaelvallat/entropy))
- [ ] Approximate entropy ([EntroPy](https://github.com/raphaelvallat/entropy))
- [ ] Lempel-Ziv complexity ([EntroPy](https://github.com/raphaelvallat/entropy))
- [ ] Fuzzy entropy ([matlab](https://www.mathworks.com/matlabcentral/fileexchange/31888-fuzzy-entropy-and-mutual-information))
- [ ] Inherent fuzzy entropy (?)
- [ ] ...

### Fractal Dimension

- [ ] detrended fluctuation analysis (DFA) ([nolds](https://github.com/CSchoel/nolds), [EntroPy](https://github.com/raphaelvallat/entropy))
- [ ] correlation dimension ([nolds](https://github.com/CSchoel/nolds))
- [ ] Lyapunov exponent ([nolds](https://github.com/CSchoel/nolds))
- [ ] Hurst exponent ([nolds](https://github.com/CSchoel/nolds))
- [ ] Petrosian fractal dimension ([EntroPy](https://github.com/raphaelvallat/entropy))
- [ ] Katz fractal dimension ([EntroPy](https://github.com/raphaelvallat/entropy))
- [ ] Higuchi fractal dimension ([EntroPy](https://github.com/raphaelvallat/entropy))
- [ ] ...

LRydin, 2020-07-08:
Small comment, I guess there is little point in separating _detrended fluctuation analysis_ (DFA)  and _multifractal detrended fluctuation analysis_ (MFDFA), as the latter should always be able to do DFA if you set the power/exponent `=2` :).

DominiqueMakowski, 2020-07-09:
Thanks for that very good point 😁 I'll leave it as is just to separate the implementations (as most of DFA implementations do not allow for MDFA ;)

closed 2021-05-22 by stale[bot]