neuropsychology/NeuroKit #53: Dicrepancy in sample entropy (SampEn) between two python implementations (nolds and entro-py); which one is correct?

DominiqueMakowski, 2019-12-21:
Trying to add Sample Entropy (SampEn). Found two (working) python implementations:

- https://github.com/ixjlyons/entro-py/blob/master/entropy.py by @ixjlyons
- https://github.com/CSchoel/nolds/blob/master/nolds/measures.py#L670 by @CSchoel

I added and adapted the former in here (currently on dev) for quick testing, but unfortunately, they give a slightly different result:

```python
signal = np.cos(np.linspace(start=0, stop=30, num=100))
nk.entropy_sample(signal, order=2, r=0.2)
0.26809106297461127
nolds.sampen(signal, emb_dim=2, tolerance=0.2)
0.2767230586620615
```

Unfortunately, I wasn't able to trace down where the discrepancy appears, and I don't which implementation is the "correct" one, as I wasn't able to compare it with other implementations.

If someone has a hint on where/why/what is the correct solution, do not hesitate ☺️ 

Other implementations:

- [entropy](https://github.com/raphaelvallat/entropy/blob/master/entropy/entropy.py#L393): gives the same result as nolds
- [pyrem](https://github.com/gilestrolab/pyrem/blob/master/src/pyrem/univariate.py#L335): doesn't seem to work (?)
- [matlab](https://physionet.org/content/sampen/1.0.0/matlab/1.1-1/#files-panel)

DominiqueMakowski, 2019-12-22:
Changed it to match nolds' and entropy's output, but it is very easy to switch between the two in case we have evidence in favour of one or the other, one can just un/comment the relevant lines:

https://github.com/neuropsychology/NeuroKit/blob/5585d62b9a88121193b61b9924f13596db4d659d/neurokit2/complexity/entropy_sample.py#L79-L84

CSchoel, 2019-12-22:
Interesting. SampEn was one of the algorithms where I was quite sure I got the implementation right, since the interim results have a clear interpretation. I will have a look into this when I find the time after the holidays.

DominiqueMakowski, 2019-12-22:
Thanks, @CSchoel! I'll let you know if I find some more info. Enjoy your holidays ☺️

DominiqueMakowski, 2019-12-23:
On top of that, pyeeg's (by @forrestbao) and pyEntrp's (by @nikdon) implementations also return different results... (and it's also the case for approximate entropy) 😕

I started mentioning here the equivalences between implementations: https://github.com/neuropsychology/NeuroKit/issues/40 (which come from [these tests]( https://github.com/neuropsychology/NeuroKit/blob/master/tests/tests_complexity.py))

Hope we'll be able to clarify the cause of the discrepancies ☺️

CSchoel, 2019-12-27:
I think I found the difference between entro-py and nolds. And (of course :stuck_out_tongue_winking_eye:  ) I think the issue lies with entro-py. I forked the repo and created a pull request:  https://github.com/ixjlyons/entro-py/pull/2. See my comments there for an explanation of what I think to be the source of the discrepancy.

My fork of entro-py produces the same result as nolds:

```python
import nolds
import entropy as e
import numpy as np

num = 100
signal = np.cos(np.linspace(start=0, stop=30, num=100))
print("entro-py", e.sampen(signal, 2, 0.2, False))
print("   nolds", nolds.sampen(signal, emb_dim=2, tolerance=0.2))
```
```
entro-py 0.27672305866206137
   nolds 0.2767230586620615
```

In the future, I might also have a look at pyeeg and pyEntrp if I can find the time.

Side note: If you are pedantic, you can also count the difference that nolds uses a `<` where entro-py uses `<=`. This should not have noticeable consequences when working with a continuous distribution of input values, but to be on the safe side I will include an optional parameter to choose between the closed and open interval in the next version of nolds. I stick with `<` as a default since I compared my results against [pracma](https://github.com/cran/pracma/blob/master/R/entropy.R) as reference implementation, which also uses `<`.

DominiqueMakowski, 2019-12-28:
👏 @CSchoel nice catch!!! Glad you found the culprit ☺️ 

Closing this as the main issue has been addressed 🎉 

@CSchoel would you mind adding yourself to the [neurokit's contributors](https://github.com/neuropsychology/NeuroKit/blob/master/AUTHORS.rst) as your work on nolds (and help) was (and will be) vital for neurokit 1 and 2 😉?

closed 2019-12-28 by DominiqueMakowski

CSchoel, 2019-12-28:
> @CSchoel would you mind adding yourself to the [neurokit's contributors](https://github.com/neuropsychology/NeuroKit/blob/master/AUTHORS.rst) as your work on nolds (and help) was (and will be) vital for neurokit 1 and 2 wink?

Will do. Thanks. :smile:

CSchoel, 2019-12-28:
For the record: I found the sources of the inconsistencies between nolds and pyeeg and pyEntropy.

* https://github.com/nikdon/pyEntropy/pull/15
* https://github.com/forrestbao/pyeeg/pull/29

It was esentially the same problem as with entro-py: The last template vector of length m must be ignored, because it has no corresponding vector of length m+1.

I actualy enjoyed this detective work. :laughing: :male_detective:

DominiqueMakowski, 2020-05-08:
Just to add on top of that, I also tested 3 R implementations of SampEn as well as the **wikipedia** python one (https://en.wikipedia.org/wiki/Sample_entropy) and they all give (sometimes very) different results... 

it's unbelievable I mean SampEn  is one of the most commonly used metrics in physiology how is it supposed to be reproducible with such discrepancies 😠 

Our current implementation is the closest to the one of the [pracma](https://github.com/cran/pracma/blob/master/R/entropy.R#L31) package (although not *exactly*).