neuropsychology/NeuroKit #58: `rsp_plot`: improvements?

DominiqueMakowski, 2019-12-27:
```python
import neurokit2 as nk

rsp = nk.rsp_simulate(duration=60, respiratory_rate=15)
signals, info = nk.rsp_process(rsp)
nk.rsp_plot(signals)
```
[image]

I suppose some improvements could be made here, for instance, different colours for signals, titles, title spacing, plotting both the raw and cleaned signal etc. 

@zen-juen care to transfer your ggplot skills and taste 😏 ?

zen-juen, 2019-12-30:
[image]

Signals now have different colours, with bow raw and cleaned signal (this data in particular doesnt have alot of noise so the raw data doesn't contrast that greatly with the cleaned one). All are plotted against time (s) and have checked that it works without amplitude data :) 
Have also plotted the means for better visualization.
Titles and axis labels subject to change :P

JohnDoenut, 2019-12-30:
Looks nice! :)

I was wondering if we should maybe move the following part to https://github.com/neuropsychology/NeuroKit/blob/dev/neurokit2/rsp/rsp_process.py

https://github.com/neuropsychology/NeuroKit/blob/4d72829a75708636f8d2b80af5dc6a5a39e282dd/neurokit2/rsp/rsp_plot.py#L30-L34

Then `rsp_plot()` would stay as parameter-free as possible (in the spirit of providing a convenince function). Otherwise people would have to figure out how long their recording is.

DominiqueMakowski, 2019-12-31:
> Otherwise people would have to figure out how long their recording is.

How so? here the time axis is just "for convenience" (to have an idea of scale) and is inferred from the sampling rate and the length. The user doesn't have to input any time info. But I agree that we might also want to display the number of samples in case for instance someone quickly wants to have an idea of at what index there is a particular artefact of something, then having indices is more useful than time.

What about a double x-axis 😏 ?

> I was wondering if we should maybe move the following part to rsp_process

You mean returning a new column "Time" alongside the signals? Or having the time as the index of the dataframe? For the latter, I'm not too sure as then it makes it complicated to index (select portions of the df, locate events etc). Here the time axis was more for convenience and readability (hence in rsp_plot) than actual usage

@zen-juen for the title, maybe a general title (above all the subplots) in bold saying "Respiration (RSP) Processing". I could imagine people creating these plots for rsp, ecg etc., so it'd be nice to be able to identify what's what in a glance. 

First plot: maybe something like "Raw and Cleaned RPS" and, in the legend, that red points correspond to peaks and yellow to troughs?
 
Also, maybe mention the unit for rate and amplitude (`Breathing rate (inhales per minutes)`)?

Finally, could we change the gradient starting from like blue (darker for cleaned signal) and then segueing into green 😬 😬 Apologies for that nitpicking but in my weird mind, ECG is red-orange, RSP is blue-green, and EDA is like purple-pink. And if I see something different I'm confused 😁

JohnDoenut, 2019-12-31:
>  the time axis is just "for convenience" (to have an idea of scale) and is inferred from the sampling rate and the length. The user doesn't have to input any time info.

https://github.com/neuropsychology/NeuroKit/blob/4d72829a75708636f8d2b80af5dc6a5a39e282dd/neurokit2/rsp/rsp_plot.py#L31

I don't see how the time axis (seconds) is inferred from the sampling rate and signal length. `duration` (seconds) is an argument that needs to be supplied by the user and it currently defaults to 60. I.e., irrespective of the actual duration of the signal, the time axis will go from 0 to 60 sec (with as many steps as samples in the signal).

That's why I feel that adding the time axis during `rsp_process()` would make more sense, since during `rsp_process` all the information needed to infer time (i.e., sampling rate, number of samples) is already available. Then the time axis could simply be appended as an additional column of the DataFrame as you suggested.
I agree that plotting the signal over time in seconds instead of samples is more convenient.

DominiqueMakowski, 2020-01-01:
>  don't see how the time axis (seconds) is inferred from the sampling rate and signal length

My bad, I have misread!

@JohnDoenut @zen-juen I edited so that it now can (or not) take the sampling rate, in which case it will display time. This way feels more natural and flexible to me ☺️ 

> for the units of the breathing amplitude, we could use "a.u." (i.e., [arbitratry unit]

👍

zen-juen, 2020-01-01:
Think having the sampling rate as a parameter fits in quite nicely :) 
[image]

@DominiqueMakowski See if these aesthetics are to your liking :P

commit f9b578d2 (2020-01-01): README

close #58
closed 2020-01-01 by DominiqueMakowski