neuropsychology/NeuroKit #64: Error in calculation of breathing amplitude in rsp_rate()

JohnDoenut, 2020-01-03:
https://github.com/neuropsychology/NeuroKit/blob/fe257bbb565cbf0d397e1d61c16dce06d3b64f8a/neurokit2/rsp/rsp_rate.py#L108-L111

We currently return breathing period instead of breathing amplitude from `rsp_rate()`. Also visible here

https://github.com/neuropsychology/NeuroKit/blob/master/docs/img/README_respiration.png?raw=true

Fixing this requires a bit of refactoring, since the calculcation of breathing amplitude requires both the raw breathing signal and the breathing extrema. Currently, we only have access to the extrema in `rsp_rate()`.

We could return the signal value at the extrema alongside the extrema in `rsp_findpeaks()` in order to fix this. @DominiqueMakowski, what are your thoughts?

JohnDoenut, 2020-01-03:
@DominiqueMakowski , maybe it's easiest to have a dedicated rsp_amplitude() function in the rsp module. Calculating breathing amplitude inside `rsp_rate()` is going to be somewhat awkward and clumsy.

DominiqueMakowski, 2020-01-03:
> maybe it's easiest to have a dedicated rsp_amplitude() function in the rsp module. Calculating breathing amplitude inside rsp_rate() is going to be somewhat awkward and clumsy.

Agreed, it could be more specific and its own dedicated parameters

JohnDoenut, 2020-01-03:
I'm going to fix this after the PR for the ecg module.

JohnDoenut, 2020-01-06:
Closing this after #68

closed 2020-01-06 by JohnDoenut