neuropsychology/NeuroKit #75: Multiscale Entropy (MSE)

DominiqueMakowski, 2020-01-10:
I am trying to add MSE. Current python implementations that I found are:

- [pyEntropy](https://github.com/nikdon/pyEntropy/blob/master/pyentrp/entropy.py#L182)
- [MultiscaleEntropy](https://github.com/reatank/MultiscaleEntropy/blob/master/MultiscaleEntropy/mse.py#L119)

Unfortunately, they give different results, but I'm not sure if this is reliable as they also give a different result for sample entropy 🤔 

As the implementation *seems* fairly straightforward and heavily depends on sample entropy, I was wondering detective @CSchoel if you could take a quick look to know if it made sense to you 🕵 ☺️ 

here's the code:

https://github.com/neuropsychology/NeuroKit/blob/a31ac5b79f0f1d24d4e5f2da02535f0909713436/neurokit2/complexity/entropy_multiscale.py#L55-L62

thanks 🙌

CSchoel, 2020-01-24:
I just had a look at both codebases. I think your implementation is sound. I like the truncation approach of pyEntropy more than the padding approach of MultiscaleEntropy. And at the end of the day the MSE seems to be nothing more than the regular sample entropy applied to differently scaled versions of the time series.

However, I find it strange that all examples (including the original papers) seem to use a linear progression for the scale factor. I would think that it is more natural/interesting to have an exponential increase in scale as it is done in wavelet theory. I would therefore prefer an interface that lets me choose the range of scale values directly as a list (like MulitscaleEntropy does).

Side note: I also had a look into the difference between pyEntropy and MultiscaleEntropy and again it seems to come down to the same error in the sample entropy, that we encountered several times already.

DominiqueMakowski, 2020-01-25:
@CSchoel thanks again for your critical input! It amazes me (and kind of worries me 😅) to see that there is quite a high number of published papers using these entropy indices despite the fact that a significant number of implementations seems not to converge 🙊 

>  would therefore prefer an interface that lets me choose the range of scale values directly as a list

Let me see how can I do that

DominiqueMakowski, 2020-01-25:
What do you think about this implementation:

https://github.com/neuropsychology/NeuroKit/blob/1902c2ff745925411e4889a4b71b37c1dfcfe811/neurokit2/complexity/entropy_multiscale.py#L59-L67

And also about this "definition":

https://github.com/neuropsychology/NeuroKit/blob/1902c2ff745925411e4889a4b71b37c1dfcfe811/neurokit2/complexity/entropy_multiscale.py#L22-L23

CSchoel, 2020-01-29:
> @CSchoel thanks again for your critical input! It amazes me (and kind of worries me 😅) to see that there is quite a high number of published papers using these entropy indices despite the fact that a significant number of implementations seems not to converge 🙊

I also thought about that. My explanation is that for at least 95% of the cases this error will not produce *meaningful* differences as it should only be noticeable when the sequence length is very small, so it is not easily spotted by accident. It takes the diligence of some crazy scientist (looking at you :wink: ) to actually go about comparing different implementations in detail. And I think an additional factor is that there are not that many experts for these algorithms. I would have hoped to get more issues and corrections when I published nolds, but so far I have mostly gotten questions asking for help (which is fine, but I actually don't consider myself to be an expert on the topic). :smile: 

> What do you think about this implementation:
> And also about this "definition":

I like both quite well. It seems to be the best trade-off between flexibility and usability. The only thing that I am not sure about is the default "max" case. I think there is an upper limit where scaling the signal further produces no more meaningful results. I would say you need at least 10 template vectors of length m+1 to get a result that is somewhat expressive, so `len(signal)/(m+1)/10` should be a good upper bound? It is still early in the morning for me, so correct me if I am wrong. :sweat_smile:

DominiqueMakowski, 2020-01-29:
> but I actually don't consider myself to be an expert on the topic.

And yet pretty much everybody on GitHub and beyond considers you to be one ☺️ 

> I think there is an upper limit where scaling the signal further produces no more meaningful results.

If I understand, using the length of the signal as max would not be optimal performance-wise (basically increasing the computational need without significant benefits/results change)? If that's the case, and there is some sort of plateau in the result after a given amount of scaling, wouldn't it be possible to run a small simulation to observe it and take a value that seems appropriate based on it? What if the optimal number of template vectors of length m+1 is 9, or 11 😁

> It takes the diligence of some crazy scientist (looking at you 😉 ) 

this is how I feel tho when it comes to complexity:
[image]

CSchoel, 2020-02-09:
> > I think there is an upper limit where scaling the signal further produces no more meaningful results.
> 
> If I understand, using the length of the signal as max would not be optimal performance-wise (basically increasing the computational need without significant benefits/results change)? If that's the case, and there is some sort of plateau in the result after a given amount of scaling, wouldn't it be possible to run a small simulation to observe it and take a value that seems appropriate based on it? What if the optimal number of template vectors of length m+1 is 9, or 11 :grin:

My main argument is not about performance, but about accuracy. The basic idea is this: The higher the scaling factor, the smaller the number of template vectors that can be constructed. Take  for example a sequence of length 100 with m=2:

Scaling factor 1: 100 data points => 98 template vectors of length 3
Scaling factor 2: 50 data points => 48 template vectors of length 3
Scaling factor 3: 33 data points => 23 template vectors of length 3
...
Scaling factor 10: 10 data points => 8 template vectors of length 3
...
Scaling factor 100: 1 data point => 0 template vectors of length 3

Since the sample entropy is a conditional probability that is estimated from the data, the error in the estimation will be larger the less data we have. I would argue doing statistics with a sample size of 10 is not reasonable science anymore. As I said I am not familiar with multiscale entropy, but I would expect that using the length of the original input data as maximum scale (i.e. using 100 as the max scaling factor in the above example) would mean that users that call the function with default parameters will see some kind of pattern in the evolution of the sample entropy across the scales that then gets more and more distorted by errors as the scaling factor increases beyond some reasonable point. Depending on the implementation of the sample entropy you will also get NaNs and/or infinities in the results.

A user who knows what they are doing should be able to evaluate the full range of possible scaling factors, but I think you can prevent a lot of confusion for novices by stopping the evaluation early. The point where to stop is of course debatable and will depend on the data, but I think we have to make an educated guess here.

In nolds I used the aforementioned rule of thumb "statistics with n<10 is not science anymore". :laughing: In this case we get `n/s - m` template vectors where `n` is the length of the initial data and `s` is the scaling factor. That would mean we want `n/s - m >= 10` which gives us `s <= n / (m + 10)`. So I think `len(signal) / (m + 10)` would be a reasonable default value in the function. (My initial formula was wrong, since it did not allow template vectors to overlap.)

DominiqueMakowski, 2020-02-10:
Gotcha! Will add this as a default 👍

CSchoel, 2020-02-10:
Correct. :smiley: :+1:

DominiqueMakowski, 2020-02-10:
Thanks a lot ☺️

closed 2020-02-10 by DominiqueMakowski

CSchoel, 2020-02-10:
No Problem. :smile: