neuropsychology/NeuroKit #78: Simulation study to identify the best preprocessing pipeline for RSP and ECG

DominiqueMakowski, 2020-01-10:
>  in the meantime we could already think about a testing framework akin to wfdb, maybe using ecg_simulate() instead of a database such as MITDB or GUDB. In any case, would be nice to have some principled way of comparing different detectors without depending on third party databases.

@JohnDoenut Related to that, so last week I got hyped with the simulation functions and the preprocessing tools that were added. And I was also looking at the different sets of defaults (#48) existing in other software, and I was once again struck with the discrepancies in the preprocessing parameters suggested, i.e., what type of filter, what order, what frequency bands to filter, detrending before filtering or after, what type of detrending, etc. etc.

Although this lack of standard, well established and validated routines are common neurophysiology (and even worse in EEG), this always annoyed me a bit not to have some kind of demonstration showing that a butterworth 2nd order filter is better than a bessel 9th order with linear detrending. 

I thought, well, we have functions to simulate signals, as well as functions to distort them (see `signal_distord()` that can now also add artefacts and powerline noise), in a controlled fashion. And (at work) I have access to a server that can run some code for an extended period of time without burning up (in theory). 

So why not run a brute-force simulation study to basically test different preprocessing parameters in a (very) large number of situations.

And from there I started struggling a lot with the [simulation code](https://github.com/neuropsychology/NeuroKit/blob/master/scripts/study_preprocessing_parameters/data_simulation.py), at first by having all combinations of preprocessing parameters, but it was just too much. So I then tried to narrow it down, starting with the filtering parameters (for RSP only). 

Basically, the idea is to generate iteratively a large number of signals of different possible rates for healthy humans), then to distort them by adding a random amount of noise of random frequencies, then to clean it using given parameters and 1) compare the cleaned signal to the original signal and 2) compare the rate extracted from the cleaned signal to the rate extracted from the original signal. 

Then, basically look at the best set of parameters. This morning I tested the simulation code by generating 100 iterations (the plots are shown [on the README](https://github.com/neuropsychology/NeuroKit/tree/master/scripts/study_preprocessing_parameters)). Next week I can try with running a lot more.

But in general, the goal would be to have a working framework for testing stuff. Moreover, aside from the theoretical justifications of the parameters, having some visual "evidence" would be quite useful for the users and could make for a straightforward but appreciated study. 

What do you think?

JohnDoenut, 2020-01-10:
> I was once again struck with the discrepancies in the preprocessing parameters suggested, i.e., what type of filter, what order, what frequency bands to filter, detrending before filtering or after, [...]

I agree. Also, it seems that at least in the development of R-peak detectors, a majority of people rely on the [mitdb] for algorithm development and evaluation This is problematic for a number of reasons (overfitting of algorithm parameters to a specific dataset being the most salient one). Moreover, the way detectors are evaluated is rarely well described and transparent and it seems that these unclarities
result in a lot of falsely inflated performance claims ([Porr, 2019](https://www.biorxiv.org/content/10.1101/722397v3.full)).

I'm not sure how sensible it is to investigate preprocessing in isolation. The preprocessing requirements of different peak detection algorithms are unique and what works for say Pan-Tompkins might not work for a Wavelet detector. That is to say that I think that the preprocessing is an integral part of the peak detection algorithm itself. Therefore it might make more sense to directly evaluate the algorithm (including their idiosyncratic preprocessing)? 

Assuming that we would evaluate algorithms, here are a couple of thoughts on the brute-force search.

1. Do we want to _evaluate_ or _optimize_ algorithms (i.e., peak detectors)?
    I.e., do we a) take some existing algorithms with fixed parameters and see how
    they perform in different noise conditions or b) do we aim at optimizing the
    parameters of these algorithms by evaluating which set of parameters is the
    best trade-off (between sensitivity and precision) in the sense that it performs well in all kinds of noise conditions? I feel that it would be sensible to start with a).
    
2. What types of noise are used to distort the signal?
    Do we choose a set of noise "types"? E.g., perform the evaluation on "noise free",
    "medium noise", "severe noise" (each with a well defined characteristics)?

3. Which algorithms are evaluated?

4. What are the performance metrics for algorithm evaluation? 
    _Sensitivity_ (how many true peaks are detected?) and _precision_ (out off all
    detected peaks, how many are true peaks?) seem to be fairly [established](https://www.hindawi.com/journals/jhe/2018/9050812/).

5. What is the ground truth for performance evaluation?
    Average rate won't be sufficient to evaluate sensitivity and precision. We 
    could use the peaks detected in the completely noise free signal as ground
    truth and then see how many of those are still detected as we add more noise.

6. Does the computation time factor into the performance evaluation?
I.e., do we capture the time it takes for a given algorithm to detect all peaks? These can vary quite substantially from what I've seen.

JohnDoenut, 2020-01-10:
Also, as an aside, this is what I found to be a nice way to set up a search over a parameter space (instead of nested for loops):

`smoothwindows = [0.1, 0.125, 0.15, 0.175]
avgwindows = [0.75, 1, 1.25, 1.5]
gradthreshweights = [1, 1.25, 1.5, 1.75]
minlenweights = [0.3, 0.4]

params = [smoothwindows,
          avgwindows,
          gradthreshweights,
          minlenweights]

params_combis = list(itertools.product(*params))

results = []

combi_counter = 1

for params_combi in params_combis:

    smoothwindow = params_combi[0]
    avgwindow = params_combi[1]
    gradthreshweight = params_combi[2]
    minlenweight = params_combi[3]
`

DominiqueMakowski, 2020-01-10:
> I'm not sure how sensible it is to investigate preprocessing in isolation.

Agreed, in fact I wanted to do all the processing necessary until the rate extraction (but started with cleaning because you need to start somewhere), but I agree that it doesn't make sense alone. For instance, for RSP, I'd be curious to know what's the effect of modulating the threshold for outliers in the peaks.

> Do we want to evaluate or optimize algorithms (i.e., peak detectors)?

It seems indeed logical to start with a) and then from there iterate to optimize the best of them.

> What is the ground truth for performance evaluation?

In the example code that I wrote, I was comparing the rate signal from the processed peaks by subtracting it from the rate signal based on the original signal (and then taking the mean of the difference). But indeed they might be better ways! Although taking the number of peaks seems to not really account for their position? For instance, some strong filters might distort the peaks, resulting in a "correct" mount of true peaks detected, but their location a bit dodged.

> Does the computation time factor into the performance evaluation?

Getting the time seems fairly straightforward, although I don't think it should be a criterion at this stage. Efficiency mostly depends on the language and the implementation, and python is not the fastest language, and we (at least I 😁) tend to prefer code that is readable and clear than fast. For example, in julia where there is a lot of room for performance (and which can be very sleek and neat looking), people tend to get so obsessed with efficiency that the code gets super unfriendly and inaccessible <sub>(which might be IMO one of the reason why julia doesn't seem to be able to really take off but that's another topic)</sub>. But I'm sure once we let say find out what is the best algorithm, some people with adequate skills will reimplement it in C or something to be super fast.

> Also, as an aside, this is what I found to be a nice way to set up a search over a parameter space (instead of nested for loops):

Indeed, it's tidier than nested loops! Feel free to improve/add-on to [this folder](https://github.com/neuropsychology/NeuroKit/tree/master/scripts/study_preprocessing_parameters). Next week I can run it on a larger scale on the server.

I'd say we start with RSP, as it is fairly simpler than ECG (less steps and possibilities), and that will be the occasion to test out the best testing framework (it's like *testception* here!)

DominiqueMakowski, 2020-06-07:
closing this in favour of #245

closed 2020-06-07 by DominiqueMakowski