neuropsychology/NeuroKit #93: Resources: EMG processing and features

DominiqueMakowski, 2020-01-21:
- [Reaz et al., 2006](https://link.springer.com/content/pdf/10.1251/bpo115.pdf)
- [Farfan et al., 2010](https://link.springer.com/article/10.1186/1475-925X-9-72)
- [Merletti et al., 2010](http://www.dl.begellhouse.com/en/journals/4b27cbfc562e21b8,606681d65e2ccb94,45714f720ecf76f5.html)
- [Zhang et al., 2013](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC3514830/)
- [Wang et al., 2013](https://pdfs.semanticscholar.org/c348/f06416aa03154c40bb82d5b26cef40e5840d.pdf)

zen-juen, 2020-02-08:
`emg_plot` now marks onsets, offsets, and regions of activity greater than the specified threshold!
[image]

DominiqueMakowski, 2020-02-08:
Aesthetic suggestions: Maybe instead of points for onsets/offsets, adding back grey vertical dashed lines behind would better represent the activation phases 😅

zen-juen, 2020-02-08:
Hmmmmm
[image]

DominiqueMakowski, 2020-02-08:
😬 we can mark onsets and offsets the same colour in the plot ('cause it's pretty obvious which one is which). And maybe plotting the vertical lines behind? As for the legend, I'd only put the orange line saying "Amplitude", the rest is rather explicit (?)

zen-juen, 2020-02-08:
🗡🗡🗡🗡🗡🗡🗡

zen-juen, 2020-02-09:
[image]

commit a562fb25 (2020-02-08): shade regions

Close #93
closed 2020-02-11 by DominiqueMakowski