neuropsychology/NeuroKit #100: RSP_clean run time and memory error

Mitchellb16, 2020-01-23:
New user here, trying to implement this package with my rodent whole body [plethysmography](https://github.com/Mitchellb16/biof509-final/blob/master/BIOF309_Final_MNB_2019.ipynb). The processing methods I've used prior are rudimentary and written up totally on my own, and your package utilizes much more robust methods.

## The issue
I'm not sure if this is just a problem on my end, but when running this:

`rsp_cleaned = nk.rsp_clean(raw_signal_vals[:20000],sampling_rate=5000, method = 'khodadad')`

the command has quite a large run time (>10 min on i7 processor w/ 16GB RAM) and the signal I passed is just a fraction of the actual file (118,000,000 samples, about 6.5 hours of recording at 5khz). When I run it on the whole signal, the kernel crashes due to a memory error. Is my data simply too much for these methods to handle, or is there something I can do to help here?  

*Note: The signal is imported from CEDSpike2 files using [NeoIO](https://neo.readthedocs.io/en/stable/io.html) and comes out as a numpy array.

Mitchellb16, 2020-01-23:
Ah, it seems the array output by NeoIO has dimensions (n, 1) and rsp_clean needs a flat 1D array. My apologies for an overly-simple 100th issue!

closed 2020-01-23 by Mitchellb16

DominiqueMakowski, 2020-01-24:
@Mitchellb16 congrats on being the winner of the 100th issue 😁