neuropsychology/NeuroKit #104: rsp_clean for high frequency (rodent) respiration

Mitchellb16, 2020-01-24:
**High frequency breathing like that done by rodents during sniffing is flattened out by the filters used in both cleaning methods.**
This may be outside the scope of the aims of this package, but if we could extend the rsp_clean function to be able to handle a range of respiratory frequencies it would be very useful to my lab, I'd love to contribute here if you're interested. 

Below is an example of the Khodadad filter flattening out the high frequency sniffing respiratory cycles of a rat. Rats usually breath in the range of 85-100 cpm, **but during sniffing can get as high as 600 cpm**. A very similar, in fact more drastic, effect is seen with the biosppy filter. 

[image]

**How could we do it?**
I'm still looking into the details of how these filters work, but perhaps we could pass an optional boolean parameter to the clean function (e.g. rodent = true or something) that changes the highcut value in [signal_filter](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/signal/signal_filter.py) from the default value of 2 to something more appropriate for the animal (~10). On that note we could even extend this to a range of animals and have default high and lowcut values for a range of typical model organisms.

JohnDoenut, 2020-01-24:
@Mitchellb16 , thank you for opening the issue. That's interesting, I haven't considered non-human respiration so far. I believe that in your case the quickest solution would be to stitch together your own pipeline:

```python
import neurokit2 as nk
import matplotlib.pyplot as plt

# Simulate 5 seconds of breathing data with a rate of 600 bpm.
rsp = nk.rsp_simulate(duration=5, sampling_rate=250, respiratory_rate=600)

# Detrend the data and subsequently apply a lowpass-filter.
rsp_detrended = nk.signal_detrend(rsp)
rsp_filtered = nk.signal_filter(rsp_detrended, highcut=10, sampling_rate=250)

# Get the breathing extrema.
_, info = nk.rsp_peaks(rsp_filtered, sampling_rate=250)

# Plot everything.
fig, ax = plt.subplots()
ax.plot(rsp, label="raw respiration")
ax.plot(rsp_filtered, label="filtered respiration")
ax.scatter(info["RSP_Peaks"], rsp[info["RSP_Peaks"]],
           label="inhalation peaks", c="r")
ax.scatter(info["RSP_Troughs"], rsp[info["RSP_Troughs"]],
           label="exhalation troughs", c="m")
ax.legend(loc="upper right")
```

This isn't as convenient as calling `rsp_process`, but it does the job in a couple of lines.

@DominiqueMakowski, I feel like this is a nice reminder for us to start documenting sooner rather than later :)

DominiqueMakowski, 2020-01-25:
> I feel like this is a nice reminder for us to start documenting sooner rather than later :)

Oh yes, that becomes a priority now ☺️ 

@Mitchellb16, I think that your issue might be the occasion for a good applied example on how to build custom pipelines. Would you like to help us write such small example (similar to [this one](https://neurokit2.readthedocs.io/en/latest/examples/qrs_extraction.html)? And also, is it possible for you to share a small portion (like 10 seconds or something) of a signal in .csv format so we can use it in the example? Thanks!

Mitchellb16, 2020-01-25:
@DominiqueMakowski , I'd be happy to help with the pipeline writing, and can also share a .csv, but won't have access to the raw data file until Monday.

DominiqueMakowski, 2020-01-25:
Don't worry, we don't have any deadline or time pressure ☺️ You can upload the data segment to [/data/](https://github.com/neuropsychology/NeuroKit/tree/master/data) (maybe something like `example_rsp_rodent.csv`.

Also, I have added some steps on how to help out with documentation (https://github.com/neuropsychology/NeuroKit/issues/87). Could you guys let me know if it's clear, understandable and detailed enough? Thanks a lot

DominiqueMakowski, 2020-03-19:
I am closing this for now as the main object of the issue has been addressed ☺️

closed 2020-03-19 by DominiqueMakowski