neuropsychology/NeuroKit #108: Return separate figures for epochs and continuous signals from ecg_plot()

JohnDoenut, 2020-02-05:
@Tam-Pham , @zen-juen , @DominiqueMakowski , currently `ecg_plot()` plots the epochs against time, whereas the continuous signals in the two subplots above can either be plotted against time or samples depending on whether or not the `sampling_rate` is specified. Maybe we can return a separate figure for the epochs? Then the x-axis labels cannot be confused across the subplots. Also, `ecg_plot()` is broken since fc6437c6635bf794c7b5b5e5fcabb9b0bdc6772b.

zen-juen, 2020-02-06:
Hi @JohnDoenut, this issue is now fixed in `ecg_plot`. So if `sampling rate=None`, only the *Raw and Cleaned Signal* and *Heart Rate* subplots are produced, with the x-axis indicating the number of samples. 

```python
nk.ecg_plot(ecg_signals, sampling_rate=None)
```
[image]

On the other hand, if it is specified i.e., `sampling_rate=1000`, the *Individual Heart Beats* subplot is additionally generated, with x-axis equal to time in seconds across all subplots in the figure.

```python
nk.ecg_plot(ecg_signals, sampling_rate=1000)
```
[image]

JohnDoenut, 2020-02-06:
Looks awesome! I like that the x-axes of the continuous signals and epoch are now clearly separated, yet we have all the information in one figure, nice solution. Closing the issue,

closed 2020-02-06 by JohnDoenut