neuropsychology/NeuroKit #111: signal_erp(): a basis for event related analysis

DominiqueMakowski, 2020-02-07:
So now that we the processing of raw signals is getting in shape, the last piece of the short-term puzzle is the implementation of functions to facilitate event-related analysis. Meaning, the extraction of relevant features for each event (obtained for instance via `epochs_create()`). 

Before going into the implementation of `ecg_erp`, `eda_erp` etc., we could have a generic basis in `signal_erp()` that can extract from epochs things like maximum, minimum, time of max, time of min, presence of peak or not, etc. etc.

What would the most appropriate name for this set of functions? `_erp()`? `_eventrelated()`, `_events()`?

zen-juen, 2020-02-18:
The **generic** `signal_erp()` produces the max, min and their corresponding time points of the signal (each index represents the epoch).
```
 # Simulate signal
 signal = nk.signal_simulate(duration=10, frequency=1)
 events = nk.signal_findpeaks(signal)
 
 # Create epochs
 epochs = nk.epochs_create(signal, events=events["Peaks"], epochs_duration=5, epochs_start=-0.1)
 nk.signal_eventrelated(epochs)

 Maximum   Minimum  Time of Maximum  Time of Minimum
1   0.500000 -0.500000          0.00002          0.50012
2   0.500000 -0.500000          0.00002          0.50012
3   0.500000 -0.499999          0.00002          0.50012
4   0.499999 -0.500000          3.99982          4.49992
5   0.500000 -0.500000          3.99982          4.49992
6   0.500000 -0.500000          4.00082          4.50092
7   0.500000 -0.500000          3.00062          3.50072
8   0.500000 -0.500000          2.00042          2.50052
9   0.500000 -0.500000          1.00022          1.50032
10  0.500000 -0.500000          0.00002          0.50012
```

For the **specific** functions, 
`rsp_eventrelated()` produces the mean rsp rate, max, min and mean of rsp amplitude (all adjusted for baseline), and whether the onset of the event concurs with rsp inspiration (coded 1) or expiration (coded 0):
```
Mean_RSP_Rate  Max_RSP_Amplitude  ...  Mean_RSP_Amplitude  RSP_Inspiration
1      -0.198968           0.009500  ...           -0.008409              0.0
2       1.099795           0.063590  ...            0.031010              0.0
3      -0.380902           0.026257  ...            0.013117              0.0
4       1.629826           0.022563  ...            0.012979              1.0
```
`ecg_eventrelated()` produces just the mean and min ecg rate, both adjusted for baseline:
```
  Mean_ECG_Rate  Min_ECG_Rate
1      -1.036617     -2.586870
2      -3.958760     -7.250912
3       2.781527     -0.272003
4      -2.071263     -5.484756
```
`emg_eventrelated()` produces the mean and max emg amplitude (not adjusted for baseline) as well as whether there is activation following the event (coded 1 if EMG onsets concur immediately after t > 0)
```
Mean_EMG_Amplitude  Max_EMG_Amplitude  Presence_of_Activation
1            0.138647           0.349223                     1.0
2            0.133452           0.349223                     1.0
3            0.090078           0.326395                     1.0
```
`eda_eventrelated()` detects whether there is activation following the event (coded 1 if SCR onsets are present within the epoch) and if yes, its corresponding peak amplitude, time of peak, rise time, and recovery time (otherwise NA elsewhere)
```
Presence_of_Activation Peak_Amplitude Time_of_Peak Rise_Time Recovery_Time
1                       1        3.11481      4.71688      1.74            NA
2                       0             NA           NA        NA            NA
3                       0             NA           NA        NA            NA
4                       1        1.67592      2.84421      1.73            NA
```

zen-juen, 2020-02-18:
PS. shout out here if you think of better column names :D (or other event related features)

DominiqueMakowski, 2020-02-20:
bad idea in the end, let's close this

closed 2020-02-20 by DominiqueMakowski