neuropsychology/NeuroKit #113: smarter bio_process()

DominiqueMakowski, 2020-02-07:
if the first argument is a dataframe, try to recover the relevant signals from the its columns (ecg will be the column that has "ECG" or "EKG" in the name, etc.)

zen-juen, 2020-02-09:
Included a `data=None` argument to contain different types of signals in one dataframe and detect them as such:
```
    if data is not None:
        if "ECG" in data.keys():
            ecg = data["ECG"]
        else:
            ecg = None
        if "RSP" in data.keys():
            rsp = data["RSP"]
        else:
            rsp = None
        if "EDA" in data.keys():
            eda = data["EDA"]
        else:
            eda = None
        if "EMG" in data.keys():
            emg = data["EMG"]
        else:
            emg = None
        
        # To store in 'keep' e.g. photosensor signal
        cols = ["ECG", "RSP", "EDA", "EMG"]
        keep_keys = [key for key in data.keys() if key not in cols]
        if len(keep_keys) != 0:
            keep = data[keep_keys]
        else:
            keep = None
```

DominiqueMakowski, 2020-02-09:
The user has to know that it is possible to pass a df instead of individual signals, but if he knows it he can do the latter, hence an additional argument is not needed. 

What I had in mind was rather a safey net for wrong input types, say the user doesn't take time to check the function args, and he just runs `bio_process(data)`, so here the dataframe is passed as the first argument (`ecg`). 

Also, rather trying to extract the column named exactly "ECG" for instance, it would be better to attempt to locate a column *containing* "ecg" or "ekg", and if none is found, set the argument to None.

Another case is that instead of passing a proper vector for the signals, the user passes a dataframe with only one column (which would appear similar to a beginner), so we want to detect if that's the case and extract this column as a vector.

commit f5457091 (2020-02-10): Update bio_process.py

fix #113
closed 2020-02-10 by zen-juen