neuropsychology/NeuroKit #117: EDA decomposition: SCRanalyze and Ledalab

DominiqueMakowski, 2020-02-09:
Currently, we have cvxEDA and Acqknowledge as methods for EDA decomposition, and plan to have the update by Amin (2019) #36 

For the sake of completeness, would be good to have the two others, namely [Ledalab](https://www.sciencedirect.com/science/article/pii/S0165027010002335) and SCRanalyze (the latter being deemed as superior by [Bach, 2014](https://www.sciencedirect.com/science/article/pii/S0301051114001847), thus a priority).

Existing implementations:

- [pyphysio](https://github.com/MPBA/pyphysio/blob/master/pyphysio/estimators/Estimators.py#L191) *(Ledalab - python adaptation)*
- [Pypsy](https://github.com/brennon/Pypsy) *(Ledalab - python adaptation)*
- [PsPM](https://github.com/bachlab/PsPM) *(SCRanalyze - original matlab code)*

hcp4715, 2020-04-02:
Hi, there, 
Just FYI, there will be an online workshop about SCR data from Bach's Lab:
https://bachlab.github.io/PsPM/courses/
I will be mainly about PsPM

DominiqueMakowski, 2020-04-02:
Thanks @hcp4715, I hope they'll show how to implement it in python 😬

closed 2021-05-22 by stale[bot]