neuropsychology/NeuroKit #120: ecg_peaks_plot(): expose the artifacts plotting

DominiqueMakowski, 2020-02-12:
Or `ecg_fixpeaks_plot()`. @JohnDoenut should we expose the plotting function?

JohnDoenut, 2020-02-12:
@DominiqueMakowski, do you mean a separate ecg plotting function other than `ecg_plot()`? In that case I don't believe that exposing the artifact plotting in `ecg_fixpeaks()` would add a lot of value. A user would need to compute the artifacts before plotting them anyway. I.e., they might as well set `show=True` _while_ computing the artifacts. On the other hand I could see some merit in generating an artifact plot from within the existing `ecg_plot()`. Of course it might be that I'm missing another use case?

DominiqueMakowski, 2020-02-13:
Good point. Do you have in mind a separate panel/line with the existing plot, or simply, for instance, marking artefacts in the main signal plot (for instance with a different colour)?

JohnDoenut, 2020-02-14:
Yes, either a) returning the two figures generated in `ecg_fixpeaks()` from `ecg_plot()` (the latter would then return a total of three figures) or, b) marking the artifact types in the first subplot of the figure that is currently returned from `ecg_plot()`. Option a) seems more informative to me.

DominiqueMakowski, 2020-02-15:
We could try give it a try and see how it goes if we add all the plots from [`_plot_artifacts_lipponen2019`](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/ecg/ecg_fixpeaks.py#L271) to the main ecg_plot (and try to organize it neatly, as for instance with the individual heart beats on the side)

zen-juen, 2020-04-04:
Added `artifacts=True` to expose artifacts in the main `ecg_plot`, so now it produces two figures like so!
```
   import neurokit2 as nk

   ecg = nk.ecg_simulate(duration=15, sampling_rate=1000, heart_rate=80)
   signals, info = nk.ecg_process(ecg, sampling_rate=1000)
   nk.ecg_plot(signals, sampling_rate=1000, artifacts=True)
```
[image]
[image]

DominiqueMakowski, 2020-04-04:
Maybe we could rename the argument with `show_artifacts` or, maybe, `show_type="..."` (to keep the same name as @Tam-Pham put in ecg_delineate). We could have show_type="default" (only the upper part), and then if show_type=="full" or "artifacts" then show artifacts?

Or maybe I'd keep the full artifacts plotting as a show arg in ecg_fixpeaks. After all, people that want to have the diagnostic plots are likely to use the subfunctions separately anyway. However, one of the things that we could transfer over to the main plot by default is the markers (the crosses) for the artifacts (ectopic, false positive/negative etc.) that could be plotted instead of the yellow points for the R peaks.  i.e., yellow points for the R peaks, replaced with crosses of color if artifact? That wouldn't add too much visually but would add the key information (where artifacts are detected)

@JanCBrammer thoughts?

JanCBrammer, 2020-04-06:
> Or maybe we should keep the full artifacts plotting as a show arg in ecg_fixpeaks. After all, people that want to have the diagnostic plots are likely to use the subfunctions separately anyway.

+1, I find the plot above too busy / overloaded.

However, I like it that all artifact-related information is now grouped together (lower half of the plot above), I would be in favor of simply returning the lower half of the plot from `ecg_fixpeaks()`.

zen-juen, 2020-04-07:
Actually these are two separate plots (the main ecg plot with rate and heartbeats + the artifacts plot). It just looks like one whole figure when I uploaded them here haha sorry for the confusion! 

So I will keep the full artifacts plot in `ecg_fixpeaks()`. Should I leave `ecg_plot()` as it originally was then?

JanCBrammer, 2020-04-07:
>  It just looks like one whole figure when I uploaded them here haha sorry for the confusion!

Ok, my bad, I should have read what you posted more carefully :D

> so now it produces two figures like so!

How about giving people the opportunity to return the artifact plot from both `ecg_fixpeaks()` as well as `ecg_plot()`?

DominiqueMakowski, 2020-04-07:
could be done with a `show_type` argument

zen-juen, 2020-04-07:
Just done with 3 options for `show_type` which can be 'default', 'artifacts' or 'full' :)

closed 2020-04-08 by zen-juen