neuropsychology/NeuroKit #122: rsp_amplitude returns wrong values?

Mitchellb16, 2020-02-13:
I've been working to get an example/pipeline going for #104 , and ran into this issue with what seem to be the wrong values for amplitude. I looked into [rsp_amplitude](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/rsp/rsp_amplitude.py) and I think that in spite of ensuring that each trough corresponds to its adjacent peak by making their lists equal length, they may have somehow become bumped in one direction or another? I'm not sure but I have 2 examples of it below. 

In both of the time segments it seems that there is a large amplitude breath near the end of the segment, but a larger amplitude is only seen in the preceding cycles, where they should not be.  

[image]

[image]

DominiqueMakowski, 2020-02-14:
Hi @Mitchellb16!

Indeed, seems like there is a problem, thanks for reporting. Could you try updating your version of neurokit (based on the plot I suspect that it is outdated, and I recall @JohnDoenut fixing the amplitude).

Try `pip uninstall neurokit2` and then `pip install https://github.com/neuropsychology/neurokit/zipball/master`

Let me know!

JohnDoenut, 2020-02-14:
@DominiqueMakowski , I believe that @Mitchellb16  is right. I noticed what @Mitchellb16 describes some time ago as well and always meant to look into it. After some initial poking around it seems that the bug is in `signal_interpolate()`. I think that we should simplify `signal_interpolate()` since I feel that the use of `x_axis` and `new_x` is somewhat convoluted.

Mitchellb16, 2020-02-14:
Hey @DominiqueMakowski, thanks for the help on this!  It seems the problem persists after updating. However, just to add some complexity, it seems when I crop the file to the 10 seconds surrounding this problem area, the values look more accurate, suggesting the problem likely occurred somewhere earlier in the file. 

Also now that I think about it, it looks like both rate and amplitude values have been "shifted" (compare with my second example above).

Also also, regarding the outdated look of the plot, I built off of rsp_plot() so that it would take time values as a parameter, just so I could quickly look at segments of interest (while also omitting inhale/exhale shading because I didn't need them here). 

[image]

DominiqueMakowski, 2020-02-26:
@Mitchellb16 It should be fixed with the recent update ☺️

Mitchellb16, 2020-03-10:
Fixed indeed! Thanks

closed 2020-03-10 by Mitchellb16