neuropsychology/NeuroKit #124: Continuous measure of ECG signal quality

DominiqueMakowski, 2020-02-14:
It is possible to get some measure of "signal quality" for each QRS complex. NK1 did that using a machine learning model. The main trick is to get the right reference template to which the QRS complexes will be compared to.

Note that for that, @hcp4715 mentioned a new [database](https://twitter.com/ScientificData/status/1227547929981878272?s=20).

However, if we have a value of "quality for each R-peak, we could then interpolate it to get a continuous index of quality, that could be used to potentially detect noisy segments.

That being said, I wonder if they are any existing and validated methods that have been published or implemented out there.

DominiqueMakowski, 2020-03-29:
Now that we have `ecg_quality()` we should 1) add it to `ecg_process()` and 2) add it to `ecg_plot()`:

I think we could try plotting a green area in the background, but very very light (low alpha), of the top subplot (containing the raw & cleaned ECG signal).

closed 2020-03-29 by DominiqueMakowski

DominiqueMakowski, 2021-01-22:
Hi @TiagoTostas, just saw your [PR](https://github.com/PIA-Group/BioSPPy/pull/75) with the signal quality function, great job 👏 Would you agree and be interested if we/you added it also here in NK ☺️?

DominiqueMakowski, 2021-01-23:
@TiagoTostas looking at the implementation, it seems like it requires two R peaks detectors. But is this method appropriate in the case when one only has one detector / one set of detected R peaks?

TiagoTostas, 2021-01-24:
This SQI takes makes use of four different quality indexes. Each one of the four indexes is classified as either optimal, suspicious or unqualified and these are then used to classify the signal according to the following function:
[image]

Using one detector makes it impossible to extract one of the features, the matching degree between the QRS, sSQI. You could adjust the decision rule and use only the other indexes three indexes (power of baseline, power of QRS and kurtosis of the signal). Any particular reason for that?

DominiqueMakowski, 2021-01-25:
Alright I should read the paper first 😁

commit 9164400b (2020-03-29): add `ecg_quality` for relative quality

close #124