neuropsychology/NeuroKit #126: rsp_simulate: smooth it to look more natural

DominiqueMakowski, 2020-02-17:
Although respiration simulation currently uses a complex model that generates pauses in inspiration and expirations, it doesn't look very natural (the breaks between are very sharp). Hence, I wonder if we should apply a smoother on top of it to make it look more natural.

zen-juen, 2020-02-21:
`signal_smooth()` added to reduce the sharpness of the breaks between inhalation and exhalation, now visualized as such: 
```
simulated_respiration = signal_smooth(simulated_respiration,
                                       kernel='boxzen',
                                       size=sampling_rate/2)
```

[image]

zen-juen, 2020-03-10:
@DominiqueMakowski can this be closed pls? 😄

DominiqueMakowski, 2020-03-10:
Can lah, for next issues if you make a specific commit that closes a given issue, you can write in the description of the commit `close #126` and it will automatically close the issue (and show which commit closed it) when the PR is merged 😉

closed 2020-03-10 by DominiqueMakowski