neuropsychology/NeuroKit #127: ECG delineation using machine/deep learning?

DominiqueMakowski, 2020-02-20:
We are currently implementing methods to delineate QRS segments (i.e., identifying the other peaks, such as the T wave, the P wave, and their offsets/onsets). Aside from the methods that we are currently adding, we've been wondering about a ML approach to it. 

The good thing is that we can easily generate tons and tons of data, using `ecg_simulate()` that generates a realistic (yet with some variability) signal.

We can 1) generate a lot of signal, 2) delineate the QRS components using the existing methods, which should work great on such clean (and "perfect") signal. This would be the "true" location of the targets. 3) We can then randomly alter the signals using `signal_distord()` to add some random noise to it, and finally 4) train the algorithm on the distorted signals to retrieve the "true" targets.

@hungpham2511 what would be the most adapted method for that? Would it be feasible with like scikit-learn or will we need to go with tensorflow? Would it work by directly feeding the whole signal in or do we have to think about meaningful features to extract?

duylp, 2020-02-21:
That's an interesting idea. Thanks to @Tam-Pham explanation of the QRS, I think it sounds like an 1D version of the popular image segmentation problem.

> @hungpham2511 @duylp what would be the most adapted method for that? Would it be feasible with like scikit-learn or will we need to go with tensorflow?

First, I think we should gather a dataset and standardize the data format. High quality data is crucial for tasks like this. Scikit-learn is a good start for simple model and then we move to tensorflow/pytorch for complex ones

> Would it work by directly feeding the whole signal in or do we have to think about meaningful features to extract?

If we use neural nets then yes, the nets will do the feature engineering.

DominiqueMakowski, 2020-03-30:
closing this in favour if #89

closed 2020-03-30 by DominiqueMakowski