neuropsychology/NeuroKit #128: ECG: Respiratory Sinus Arrhythmia (RSA)

DominiqueMakowski, 2020-02-20:
Need to implement:

- [ ] Respiration from ECG
  - [Kim, 2006](https://link.springer.com/chapter/10.1007/978-3-540-36841-0_1030)
  - [Charlton, 2016](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC5390977/)
  - [Helfenbein, 2014](https://www.ncbi.nlm.nih.gov/pubmed/25194875)
  - [Sarkar, 2015](https://www.researchgate.net/publication/304221962_Extraction_of_respiration_signal_from_ECG_for_respiratory_rate_estimation)
- [ ] P2T (peak-to-trough) method
- [ ] PB (Porges-Bohrer) method
- [ ] High frequency HRV
- [ ] (?)

References:
- [Lewis, G. F., Furman, S. A., McCool, M. F., & Porges, S. W. (2012). Statistical strategies to quantify respiratory sinus arrhythmia: Are commonly used metrics equivalent?. Biological psychology, 89(2), 349-364.](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC3269511/)
- [Servant, D., Logier, R., Mouster, Y., & Goudemand, M. (2009). La variabilité de la fréquence cardiaque. Intérêts en psychiatrie. L'Encéphale, 35(5), 423-428. *(french)*](https://www.sciencedirect.com/science/article/abs/pii/S0013700608002649)

DominiqueMakowski, 2020-02-21:
Note: make it so that the RSA/HRV functions can accept a dataframe of processed signals from which it retrieves the rate and the rpeaks, similar to ecg_plot.

zen-juen, 2020-03-20:
For now, the function has one input argument for processed signals (assuming that rsp and ecg signals are in one dataframe, such as that produced by `bio_process()`) and outputs a dictionary of RSA metrices. Just wondering should the `signals` input be split into one for rsp and one for ecg? Also any more aesthetic way to print the metrices?  😄 
```
import neurokit as nk

data = pd.read_csv("https://raw.githubusercontent.com/neuropsychology/NeuroKit/dev/data/bio_eventrelated_100hz.csv")

# Process the data
signals, info = nk.bio_process(ecg=data["ECG"], rsp=data["RSP"], sampling_rate=100)
rsa = nk.ecg_rsa(signals, info)
rsa
```

```
{'RSA_P2T_Values': [0.009000000000000008,
  0.01100000000000001,
  0.012999999999999998,
  0.0,
  0.011999999999999997,
  0.0040000000000000036,
  0.0020000000000000018,
  0.0010000000000000009,
  0.0030000000000000027,
  0.007000000000000006,
  0.0050000000000000044,
  0.0,
  0.008999999999999994,
  0.011999999999999997,
  0.0040000000000000036,
  0.0010000000000000009,
  0.00399999999999999,
  0.0030000000000000027,
  0.0020000000000000018,
  0.018000000000000002,
  0.0050000000000000044,
  0.002999999999999989,
  0.015,
  nan,
  0.012999999999999998,
  0.006999999999999992,
  0.0049999999999999906,
  0.0020000000000000018,
  0.009999999999999995,
  0.013999999999999999,
  0.0049999999999999906,
  0.006000000000000005,
  0.006999999999999992,
  0.039999999999999994,
  0.0030000000000000027,
  0.01100000000000001,
  0.006000000000000005,
  nan,
  0.008000000000000007],
 'RSA_P2T_Mean': 0.00756756756756757,
 'RSA_P2T_Mean_log': -4.8838835884571115,
 'RSA_P2T_Variability': 0.007143530641778618,
 'RSA_P2T_NoRSA': 2,
 'RSA_PorgesBohrer': -2.378740257402858}
```

DominiqueMakowski, 2020-03-21:
> Just wondering should the signals input be split into one for rsp and one for ecg? 

Mmh it's trickier than I thought. Maybe something like this could work?:

```
ecg_rsa(ecg_signals, rsp_signals=None, ...)
```

We mention that ecg_signals has to be a dataframe containing two columns, namely the ecg_rate and ecg_R_peaks. rsp_signal has to be a dataframe containing also whatever is needed. However, the stuff for `rsp_signals` can be extracted from the dataframe provided to the first argument (so the second becomes "optional"). 

> Also any more aesthetic way to print the metrices? 😄

Unfortunately, python doesn't easily allow to tweak the printing of standard objects :) 

Also I'd put the different methods (P2T, PB) as distinct internals (similar to HRV with the different domains).

closed 2020-04-10 by DominiqueMakowski