neuropsychology/NeuroKit #131: ECG: Heart Rate Variability (HRV)

DominiqueMakowski, 2020-02-26:
## HRV

### General
- [Guidelines for Reporting Articles on Psychiatry and Heart rate variability](https://www.nature.com/articles/tp201673)
- [Slides by @michagaebler (2020) on HRV](https://osf.io/4xdmv/)

### Complexity 

- [Entropy analysis of heart rate variability and its application to recognize major depressive disorder](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC6597986/)
- [Entropy Analysis of Short-Term Heartbeat Interval Time](https://www.mdpi.com/1099-4300/19/10/568)

DominiqueMakowski, 2020-06-23:
@Tam-Pham could you take a look at these super easy formulae: https://link.springer.com/article/10.1186/s12938-017-0402-3 ^^ so we can add these additional poincaré indices as it seems highly relevant especially for short-term intervals 🙏

DominiqueMakowski, 2020-06-28:
### Time-frequency Domain Analysis (TF)

Still from https://www.ncbi.nlm.nih.gov/pmc/articles/PMC3913584/pdf/pone.0081896.pdf; 3 methods: 

- nonparametric linear TF methods based on linear filtering
  - short-time Fourier transform
  - wavelet transform
- nonparametric quadratic TF representations
  - Wigner-Ville distribution and its filtered versions (e.g., Smoothed Pseudo Wigner-Ville distribution; SPWVD)
- parametric time-varying methods
  - autoregressive models with time-varying coefficients

We might want to include some of these as generic methods under something like `signal_timefrequency` or something.

[image]

> Each HRV signal is divided into 5 segments of equal length, each segment is approximately 15 seconds in time domain. The average energy of each segment was computed. The features are:
> - MAX w: maximum amount of energy in each window.
> - MIN w: minimum amount of energy in each window.
> - DIF w: difference between maximum and minimum amount of energy between windows.
> - STD w: standard deviation between energy of time windows. The obtained signal in TF domain is also divided into three frequency segments.
> - Evlf: the total energy of signal in very low frequency band (0.003–0.04) Hz, divided by length of band (0.037).
> - Elf: the total energy of signal in low frequency band (0.04–0.15) Hz, divided by length of band (0.11).
> - Ehf: the total energy of signal in high frequency band (0.15–0.4) Hz, divided by length of band (0.25).
> - Fvlf: the average of energy signal in very low frequency band (0.04–0.003) Hz.
> - Flf: the average of energy signal in low frequency band (0.04– 0.15) Hz.
> - Fhf: the average of energy signal in high frequency band (0.15– 0.4) Hz.
> - Also, we have defined the first order derivative as a feature to show the difference between adjacent windows. This derivative is the difference between the average energy in subsequent windows. This derivative for the first window(first 15 S) was computed by the difference between this window and the last 15 seconds in the second minute. So the first order derivative feature is computed as below...

Tam-Pham, 2021-03-12:
BUMPED. need to add SDANN and SDNN index

DominiqueMakowski, 2021-10-19:
Closing in favour of https://github.com/neuropsychology/NeuroKit/projects/5

closed 2021-10-19 by DominiqueMakowski