neuropsychology/NeuroKit #132: heart rate; something's wrong with the interpolation instead of resampling

DominiqueMakowski, 2020-02-26:
I'm not sure that my recent change from resample to interpolation did good, as the first segment of data should be set at the average value and now it seems off. Must look into it tomorrow.

JohnDoenut, 2020-02-26:
The samples in `new_x` that precede or follow the first and last interpolation point on the x-axis (in this case the first and last R-peak) are set to first and last heart rate value respectively.

https://github.com/neuropsychology/NeuroKit/blob/eb6f77c2327c9532798690562ce19e6f672ad1da/neurokit2/signal/signal_interpolate.py#L67

You could change it to `fill_value=([np.mean(y_values)], [np.mean(y_values)]))` to achieve the desired behavior.

DominiqueMakowski, 2020-02-27:
@JohnDoenut right, but do you validate the use of signal_interpolate instead of signal_resample?

JohnDoenut, 2020-02-27:
@DominiqueMakowski , what do you mean by validate?

I think the culprit might be these lines:
https://github.com/neuropsychology/NeuroKit/blob/3446907a08c61007666335838061ca679fb8fecb/neurokit2/ecg/ecg_rate.py#L61-L64

The first element in `rr` is set to 0 and then replaced by the mean of `rr`. For short signals (such as the one used to generate the ECG test plot), this is problematic. Since the first element with value 0 / `sampling_rate` contributes to the mean, it might distort the mean. I never noticed because I always used longer signals for which the mean isn't as vulnarable to outliers.

Try replacing
https://github.com/neuropsychology/NeuroKit/blob/3446907a08c61007666335838061ca679fb8fecb/neurokit2/ecg/ecg_rate.py#L64

by `rr[0] = np.mean(rr[1:])`.

I won't have the opportunity to look into this myself until tonight (-7 hrs Singapore time).

DominiqueMakowski, 2020-02-27:
> what do you mean by validate?

As in should we not change it and leave `resample()` as it was before?

JohnDoenut, 2020-02-27:
> As in should we not change it and leave `resample()` as it was before?

For consistency with the `rsp` functions I would be in favor of using `signal_interpolate()` in the `ecg` module instead of `signal_resample()` (`rsp_rate()` and `rsp_amplitUde(0` also call `signal_interpolate()`). 

An added bonus is that I understand what `scipy.interpolate.interp1d()` does in contrast to `scipy.ndimage.zoom()` which I find less transparent.

JohnDoenut, 2020-02-28:
@DominiqueMakowski, I think we could close this after merging #130. Changing the indexing seems to have fixed the issue.

closed 2020-03-19 by DominiqueMakowski