neuropsychology/NeuroKit #133: RSP: Respiratory Rate Variability (RRV)

DominiqueMakowski, 2020-02-27:
Very few papers on the topic.

## Ressources

- [Soni, R., & Muniyandi, M. (2019). Breath Rate Variability: A Novel Measure to Study the Meditation Effects. International journal of yoga, 12(1), 45–54.](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC6329220/) (describes some measures)

Mitchellb16, 2020-02-28:
Very interested in helping out here!

DominiqueMakowski, 2020-02-29:
@Mitchellb16 please do please do ☺️ 

On the dev branch, I added a file for HRV: https://github.com/neuropsychology/NeuroKit/blob/dev/neurokit2/ecg/ecg_hrv.py The idea would be to do it in a similar fashion for respiration. 

Feel free to share some knowledge, literature, or start coding!

closed 2020-04-10 by DominiqueMakowski