neuropsychology/NeuroKit #134: ecg_process fails when heart_rate is 90

DominiqueMakowski, 2020-02-28:
```python
ecg90 = nk.ecg_simulate(duration=30, sampling_rate=200, heart_rate=90, random_state=42)
nk.ecg_process(ecg90)
```
```
nk.ecg_process(ecg90)
Traceback (most recent call last):

  File "<ipython-input-14-210b69211481>", line 1, in <module>
    nk.ecg_process(ecg90)

  File "D:\Dropbox\RECHERCHE\N\NeuroKit\neurokit2\ecg\ecg_process.py", line 53, in ecg_process
    instant_peaks, rpeaks, = ecg_peaks(ecg_cleaned=ecg_cleaned,

  File "D:\Dropbox\RECHERCHE\N\NeuroKit\neurokit2\ecg\ecg_peaks.py", line 64, in ecg_peaks
    rpeaks = ecg_findpeaks(ecg_cleaned, sampling_rate=sampling_rate,

  File "D:\Dropbox\RECHERCHE\N\NeuroKit\neurokit2\ecg\ecg_findpeaks.py", line 97, in ecg_findpeaks
    rpeaks = _ecg_findpeaks_neurokit(ecg_cleaned, sampling_rate,

  File "D:\Dropbox\RECHERCHE\N\NeuroKit\neurokit2\ecg\ecg_findpeaks.py", line 165, in _ecg_findpeaks_neurokit
    end_qrs = end_qrs[end_qrs > beg_qrs[0]]

IndexError: index 0 is out of bounds for axis 0 with size 0
```

DominiqueMakowski, 2020-02-28:
Seems to be caused by the low sampling rate.

JohnDoenut, 2020-02-28:
@DominiqueMakowski, could you with `nk.ecg_process(ecg90, sampling_rate=200)`?

DominiqueMakowski, 2020-02-28:
Oops 😅

closed 2020-02-28 by DominiqueMakowski

JohnDoenut, 2020-02-28:
Happens to the best 😄