neuropsychology/NeuroKit #136: signal_power(): add show argument

DominiqueMakowski, 2020-02-29:
Would be useful to add the possibility of having a plot for signal_power, especially for instant power. Basically a plot similar to what `signal_psd()` does, but with areas filled with different colours for each frequency bands.

Something like this:

[image]

from https://github.com/Aura-healthcare/hrvanalysis#plot-functions

(would be useful to double-check frequency domain HRV)

Tam-Pham, 2020-04-11:
I added a chunk of code under `_signal_power_instant` for plotting:
```
 if show is True:
        frequency_band_index = []
        for band in frequency_band:
            indexes = np.logical_and(psd["Frequency"] >= band[0], psd["Frequency"] < band[1])
            frequency_band_index.append(np.array(indexes))
        # label?
        ax = psd.plot(x="Frequency", y="Power", logy=True, title='Power Spectral Density (PSD)')
        for band_index in frequency_band_index:
        # for band_index, label in zip(frequency_band_index, label_list):
            ax.fill_between(psd["Frequency"][band_index], 0, psd["Power"][band_index])
            # ax.fill_between(frequency[band_index], 0, power[band_index], label=label)

        ax.set(xlabel="Frequency (Hz)", ylabel="Spectrum")
```
And when `show=True` the plot of the following:
```
import neurokit2 as nk

ecg = nk.ecg_simulate(duration=240, sampling_rate=1000)
ecg, info = nk.ecg_process(ecg, sampling_rate=1000)
hrv = nk.ecg_hrv(ecg, sampling_rate=1000, show=True)

```
[image]

Is this what you want?

Tam-Pham, 2020-04-11:
Currently commenting out the label as I'm not too sure what's the best way to pass the label in

DominiqueMakowski, 2020-04-11:
```python
signal = nk.rsp_simulate(duration=60)
nk.signal_psd(signal, show=True)  # It's a bit ugly I wonder why it's so jitteray

nk.signal_power(signal, frequency_band=[10, 30], show=False) #  This work
nk.signal_power(signal, frequency_band=[10, 30], show=True) #This doesn't but it should
```

I think it's best if the plotting function is an independent internal ran optionally at the end of `_signal_power_instant`. Also the labels from the dataframe outputted, and it'd be nice to remove the line and leave only that coloured area?

Tam-Pham, 2020-04-13:
Sorry for the late rep! 😆 

ok, the plot function is now added as an independent internal function. 

[image]
It looks nice without the line indeed. However, if the frequency band doesn't cover the entire signal then it looks weird:
[image]

So i'm setting it as grey now. Let me know what you think.

```
nk.signal_power(signal, frequency_band=[10, 30], show=True) #This doesn't but it should
```
The above didn't work because the frequency_band is defined as a list of integer whereas the function takes in a list of tuples or a tuple of 2 int. Anyway, I added the code to sanitize this input so now the above code should work too. 

And to why:
```
nk.signal_psd(signal, show=True)  # It's a bit ugly I wonder why it's so jitteray
```
🤷‍♀️ signal is noisy? well, I think the log is already helping with the jitteray?
[image]

DominiqueMakowski, 2020-04-13:
I think we can plot the whole PSD in light grey, and highlight in color the parts asked for in `frequency_band`. As for the jittery part, maybe we should take a look at `signal_psd()`, I'm not 100% confident that it's the best way of implementing it!

Tam-Pham, 2020-04-16:
Ok, I looked at the `scipy.signal.welch()`, our implementation and others' implementations. 

Generally, I think our implementation is alright 👍 The jittery could be normal.

Anyway, I tried our functions with a real signal (16 minutes, 4000Hz), instead of a simulated ones. 

With `logy=True`: 
`
ax = psd.plot(x="Frequency", y="Power", logy=True, title='Power Spectral Density (PSD)', color='grey')
`
[image]

With `logy=False`: 
`
ax = psd.plot(x="Frequency", y="Power", logy=False, title='Power Spectral Density (PSD)', color='grey')
`
[image]

With shorter signal for more clarity:
[image]

Just what we would expect a power-frequency plot to be. 

Instead of filling the area, we can plot it with lines only:
[image]

How do y

DominiqueMakowski, 2020-04-16:
Fill ✔️ log ✔️ (I think it's common practice to logtransform - but do double check 😁)

closed 2020-04-18 by Tam-Pham