neuropsychology/NeuroKit #142: EMG: implementation of an power-based EMG segmentation method

GanshengT, 2020-03-02:
I currently implemented an EMG segmentation and visualization method based on 'mne' package. But **I am not sure about the EMG data and corresponding methods' structure in NeuroKit**.  I suggest that I put a readable code in "Scripts" and let you integrate it into NeuroKit. This would be an example for me to learn more about the NeuroKit structure.

DominiqueMakowski, 2020-03-04:
looks great 💪 We will work to integrate it as soon as I'm back from vacation next Monday :) thanks!

DominiqueMakowski, 2020-03-10:
@GanshengT Based on your script, I have put together the function below, a bit simplified (and I've replaced some bits by existing neurokit functions) and reproducible:

```python
def _emg_activation_powerbased(emg_cleaned, sampling_rate=1000, threshold=0.75):

    # Preprocessing
    signal = np.abs(emg_cleaned)  # getting absolute value of the EMG channel

    # Parameters
    window = np.int(0.5 * sampling_rate)  # related to the duration of a movement
#    step = 100  # equivalent to resolution

    # rolling window
    energy = pd.Series(signal).rolling(window, win_type='boxcar').sum()  # [::step]

    # Get onsets
    binarized_energy = nk.signal_binarize(energy.fillna(0), threshold=energy.quantile(threshold))
    activations = nk.events_find(binarized_energy, duration_min=window)

    # Prepare Output.
    info = {"EMG_Onsets": activations["onset"],
            "EMG_Offsets": activations["onset"] + activations["duration"]}

    return binarized_energy, info

```

Which gives the following output:

```python
import neurokit2 as nk

emg_cleaned = nk.emg_simulate(duration=20, n_bursts=3)
binarized_energy, info = _emg_activation_powerbased(emg_cleaned)
nk.signal_plot([emg_cleaned, binarized_energy], standardize=True)
```
[image]

```python
nk.events_plot(info["EMG_Onsets"], emg_cleaned)
```

[image]

The tricky part seems to be the selection of default parameters (threshold and window length). 

Is that roughly what you had in mind?

GanshengT, 2020-03-11:
Great work, Dominique. That is the idea. In my opinion, `nk.events_find` is suitable for finding the onset.  Thank you for pointing out that the method might not be robust for emg from different paradigms.  Ideally, the threshold is determined according to the prior knowledge about the approximate duration ration between "activation" and "resting state". I would let you know if it is possible to estimate the default parameters to further improve this method XD.

DominiqueMakowski, 2020-03-12:
Ff you believe it has potential benefits as compared to the current activation detection method (based on amplitude), we could try adding it to the module. Although first you might wanna try looking into finding a good default for the parameters ☺️

GanshengT, 2021-04-09:
This issue has been solved a long time ago. I close it now.

closed 2021-04-09 by GanshengT