neuropsychology/NeuroKit #144: plot individual heart beats: make window depend on heart rate

DominiqueMakowski, 2020-03-04:
Currently, the window that epochs the individual heartbeats is fixed. However, for very slow/fast heart rate (or in non-human), this might be not appropriate. Hence, we might want to make it depend on the average heart rate (if available).

DominiqueMakowski, 2020-03-04:
for instance we can create a "modulator" like 

```python
m = 1 - (0.0125 * np.mean(heartrate))  # Which is equal to 0 in the case of an average heart rate of 80bpm
``` 
And then we penalize the window, for instance let's say if currently the left side is -0.4s:

```python
epoch_start = -0.4 + 0.01 * m  # The tricky part is to find a good value for the coefficient (here 0.01)
```

zen-juen, 2020-03-10:
The modulator is set as  `m = heart_rate/80` with `epochs_start = -0.3/m` and `epochs_end = 0.5/m`. The epoch window will thus be narrower (for a faster rate) or wider (for a slower rate) based on the epoch window which was originally manually set with `heart_rate=80`.

zen-juen, 2020-03-10:
However, there seems to be something wrong with the "Raw and Cleaned Signal" portion of `ecg_plot()`, with the cleaned signal looking very different from the raw - perhaps a new issue for this :P

DominiqueMakowski, 2020-03-10:
> However, there seems to be something wrong with the "Raw and Cleaned Signal" portion of ecg_plot(), with the cleaned signal looking very different from the raw - perhaps a new issue for this :P

Yup ^^

closed 2020-03-11 by zen-juen