neuropsychology/NeuroKit #146: Neurokit2 error

Yuyang01, 2020-03-10:
**Describe the bug**
I run the code in the tutorial:

`import pandas as pd
import neurokit2 as nk

# Generate 10 seconds of EMG signal (recorded at 250 samples / second)
emg = nk.emg_simulate(duration=10, sampling_rate=250, n_bursts=3)

# Process it
signals = nk.emg_process(emg, sampling_rate=250)

# Visualise the processing
nk.emg_plot(signals, sampling_rate=250)`

**I got the following error**
    nk.emg_plot(signals, sampling_rate=250)

  File "/Users/yuyang01/miniconda3/envs/py37/lib/python3.7/site-packages/neurokit2/emg/emg_plot.py", line 32, in emg_plot
    onsets = np.where(emg_signals["EMG_Onsets"] == 1)[0]

TypeError: tuple indices must be integers or slices, not str

**Specifiations (please complete the following information):**
 - Nerokit2 Version: 0.0.10

DominiqueMakowski, 2020-03-10:
This is an error in the tutorial, it should be: 

```python
import pandas as pd
import neurokit2 as nk

emg = nk.emg_simulate(duration=10, sampling_rate=250, n_bursts=3)
signals, info = nk.emg_process(emg, sampling_rate=250)

nk.emg_plot(signals, sampling_rate=250)
```

Yuyang01, 2020-03-10:
Thank you. Got it.
*Yuyang Wang*

Yuyang01, 2020-03-10:
@DominiqueMakowski , @zen-juen , 

Could you also check I got the figure plot ugly? The code is also from the tutorial:

```python
# Generate 10 seconds of EDA signal (recorded at 250 samples / second) with 2 SCR peaks
eda = nk.eda_simulate(duration=10, sampling_rate=250, n_scr=2, drift=0.01)

# Process it
signals, info = nk.eda_process(eda, sampling_rate=250)

# Visualise the processing
nk.eda_plot(signals, sampling_rate=250)
```

[image]

DominiqueMakowski, 2020-03-10:
I got it ugly too, something got broken in the recent updates - will investigate. Thanks for reporting!!

zen-juen, 2020-03-10:
@Yuyang01 It seems like our `eda_plot()` code does not adjust appropriately for the `sampling_rate` argument. For now you can set sampling rate to None for the plot to be generated properly i.e. `nk.eda_plot(signals, sampling_rate=None)` rather than sampling_rate = 250, in the mean time we'll fix the code bugs and get back to you asap!

zen-juen, 2020-03-10:
@Yuyang01 @DominiqueMakowski the `eda_plot()` problem is now fixed :) You can now try running the tutorial code in your console again.

DominiqueMakowski, 2020-03-10:
@zen-juen fixed as in on the dev branch?

zen-juen, 2020-03-10:
@DominiqueMakowski yes! oops sorry forgot to mention it’s on Dev!

DominiqueMakowski, 2020-03-10:
@zen-juen I think you forgot to push the fix ^^

Yuyang01, 2020-03-10:
@zen-juen  After set the sampling_rate to None, I got the right figure. Thank you.

DominiqueMakowski, 2020-03-10:
@Yuyang01 You can resample a given signal using the [`signal_resample()`](https://neurokit2.readthedocs.io/en/latest/functions.html#neurokit2.signal_resample):

`resampled_signal = nk.signal_resample(signal, sampling_rate=250, desired_sampling_rate=200)`

And then plot it 😉

Yuyang01, 2020-03-10:
@DominiqueMakowski  Really great signal process tool in Python I have ever seen!

closed 2020-03-12 by Yuyang01