neuropsychology/NeuroKit #149: ecg_clean(): neurokit method looks very different

zen-juen, 2020-03-11:
As you can see in this figure below, the cleaned signal and the raw signal looks drastically different from each other:
```signals, info = nk.ecg_process(ecg, sampling_rate=1000)
nk.ecg_plot(signals, sampling_rate=1000)
```
[image]

It seems like the `ecg_clean()` neurokit method, which uses signal_filter() twice, scales down the raw signal alot. Crucially, it is the second portion of the code (where method = "powerline") that does this. @JanCBrammer Just wondering if this is expected to happen?
```
ecg = nk.ecg_simulate(duration=15, sampling_rate=1000, heart_rate=80)

clean_1 = nk.signal_filter(signal=ecg,
                           sampling_rate=1000,
                           lowcut=0.5,
                           method="butterworth",
                           order=5)
clean_2 = nk.signal_filter(signal=clean_1,
                           sampling_rate=1000,
                           method="powerline")
signals = pd.DataFrame({
        "ECG_Raw": ecg,
        "ECG_Cleaned_1": clean_1,
        "ECG_CLeaned_2": clean_2})
nk.signal_plot(signals)
```
[image]

JanCBrammer, 2020-03-11:
@zen-juen, you're right, the downscaling is a byproduct of the powerline filter. I see how this is problematic for `ecg_plot()`. Maybe we can normalize the raw and clean signal before plotting? Then they are both on the same scale and it's possible to observe the effect of filtering better.

DominiqueMakowski, 2020-03-11:
We have a `standardize()` function that we could use for that

JanCBrammer, 2020-03-11:
We can also simply smooth the ECG with a moving average window of `sampling_rate / 50`. We can use `signal_smooth` for that. This way we could skip the hack with the normalization I proposed earlier. Will send a PR today.

JanCBrammer, 2020-03-11:
With #150 the plot looks like this.

```
import neurokit2 as nk
import pandas as pd

sampling_rate = 1000
ecg = nk.ecg_simulate(duration=15, sampling_rate=sampling_rate, heart_rate=80)

clean_1 = nk.signal_filter(signal=ecg,
                           sampling_rate=sampling_rate,
                           lowcut=0.5,
                           method="butterworth",
                           order=5)
clean_2 = nk.signal_filter(signal=clean_1,
                           sampling_rate=sampling_rate,
                           method="powerline")
signals = pd.DataFrame({
        "ECG_Raw": ecg,
        "ECG_Cleaned_1": clean_1,
        "ECG_CLeaned_2": clean_2})
nk.signal_plot(signals)
```

[image]

DominiqueMakowski, 2020-03-11:
I'd be worried that smothering through moving windows (whether average or median or whatnot) might jitter the location of sharp peaks (such as rpeaks) more strongly than "frequency based" filtering :/ (maybe it's not the case though)

JanCBrammer, 2020-03-11:
@DominiqueMakowski , I'm definitly out of my depth when it comes to digital signal processing (DSP), but I think that scipy.signal.filtfilt also comes down to convolving the signal with a kernel (built according to the filter coefficient b and a). By using the moving average the filter coefficients are different ones, but the application of the filter is the same (convolution).

JanCBrammer, 2020-03-11:
@DominiqueMakowski , see the latest commit in #150 which also achieves moving average filtering, this time implemented with `scipy.signal.filtfilt()`. By using this implementation we also don't have to import `signal_smooth()` in `signal_filter()`.

closed 2020-03-18 by zen-juen