neuropsychology/NeuroKit #152: How to compute the ratio of LF/HF with Neurokit

Yuyang01, 2020-03-12:
Dear Developers
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
In many articles the signal is transformed into Fourier domain for analysis, and some can compute the ratio between LF (low frequency) and HF (high frequency). 

**How could we do it?**
I don't if this feature has been developed or not, I wish to analyse my signal in this way. But I have no idea how to do it.
Thank you in advance if you could give me a hand.

DominiqueMakowski, 2020-03-12:
@Yuyang01 You are right on time, as we just started implementing this kind of tools in our functions for heart rate variability. In it, we compute LF/HF using the `nk.signal_power()` function, you can see how it's done in the code and apply it to your own signal:

https://github.com/neuropsychology/NeuroKit/blob/35be5e99dc3e7c408f2d4842c271492372d7ea32/neurokit2/ecg/ecg_hrv.py#L142-L144

https://github.com/neuropsychology/NeuroKit/blob/35be5e99dc3e7c408f2d4842c271492372d7ea32/neurokit2/ecg/ecg_hrv.py#L148

Yuyang01, 2020-03-12:
@DominiqueMakowski  It works. I leave the code (from your file) here for future readers to get start:

```python
ecg = nk.ecg_simulate(duration=240)
ecg, info = nk.ecg_process(ecg)
hrv = nk.ecg_hrv(ecg, show=True)
```
Thank you.

DominiqueMakowski, 2020-03-13:
> instant vs continuous?

Instant get the amount of power in a given frequency band. It returns a point value. On the contrary, continuous estimates the fluctuations of power in a given band throughout the signal (returning a signal-like shaped array). 

> BVP vs ECG ?

For now we don't have (yet) support for PPG/BVP. So for now, I'd recommend using other packages for that, such as for instance the excellent [systole](https://github.com/embodied-computation-group/systole), to at least get your preprocessing done. 

Once you have detected the peaks, you could potentially plug that into `ecg_hrv` to then get HRV indices.

You could also consider contributing and getting started on implementing BVP processing support 😉 

> ecg_process reports errors when I use different sampling_rate?

Yeah, it's likely due to the fact that ECG and BVP signals are very different, and that you cannot really use the ECG pipeline for BVP.

Yuyang01, 2020-03-13:
Dear @DominiqueMakowski ,

Thank you for your helpful explanation. I am glad to have your invitation to contribute the package, but as I am very new to signal processing at this stage, I would postpone this for a while.

Best regards,
Yuyang

closed 2020-03-13 by DominiqueMakowski