neuropsychology/NeuroKit #169: Changes in signal_distort introduced errors

DominiqueMakowski, 2020-03-28:
```python
nk.eda_simulate(duration=10, sampling_rate=250)
```
```
Traceback (most recent call last):

  File "<ipython-input-6-832299724ab3>", line 1, in <module>
    nk.eda_simulate(duration=10, sampling_rate=250)

  File "D:\Dropbox\RECHERCHE\N\NeuroKit\neurokit2\eda\eda_simulate.py", line 81, in eda_simulate
    eda = signal_distort(eda,

  File "D:\Dropbox\RECHERCHE\N\NeuroKit\neurokit2\signal\signal_distort.py", line 91, in signal_distort
    noise += _signal_distord_noise_multifrequency(signal,

  File "D:\Dropbox\RECHERCHE\N\NeuroKit\neurokit2\signal\signal_distort.py", line 190, in _signal_distord_noise_multifrequency
    raise ValueError(f"NeuroKit error: Please choose frequencies smaller than {nyquist}.")

ValueError: NeuroKit error: Please choose frequencies smaller than 25.0.
```

Seems to be related to the frequency cap @JanCBrammer any idea how to fix?

JanCBrammer, 2020-03-28:
@DominiqueMakowski, the error message is related to the following:

https://github.com/neuropsychology/NeuroKit/blob/1450f8f798632462f3905a8721d78d26cd9f60b7/neurokit2/signal/signal_distort.py#L186-L190

At a sampling rate of 250 the maximum frequency that can be resolved according to the above criterion is 25. However, `eda_simulate()` requests a frequency of 100:

https://github.com/neuropsychology/NeuroKit/blob/1450f8f798632462f3905a8721d78d26cd9f60b7/neurokit2/eda/eda_simulate.py#L81-L85

The solution is to either lower the requested frequency to 25, or increase the sampling rate to 1000.

Note that theoretically the Nyquist (i.e., highest frequency that could be resolved) is sampling rate / 2. However, this is [not sufficient to prevent aliasing in Python](https://www.gaussianwaves.com/2020/01/how-to-plot-fft-in-python-fft-of-basic-signals-sine-and-cosine-waves/). Hence the more conservative criterion of sampling rate / 10.

DominiqueMakowski, 2020-03-28:
This could be easily fixed by removing in `signal_distort` from requested frequencies those that are above sr/2? we could add an argument, `safe=False`, in which case it would error as it does now, otherwise it would just silently drop the impossible freqs

JanCBrammer, 2020-03-28:
Or we don't raise an error but instead just print a message for those frequencies that couldn't be resolved? Something like `f"Skipping requested noise frequency of {freq} Hz since it cannot be resolved at the sampling rate of {sampling_rate}. Increase sampling rate to {freq * 10}"`?

DominiqueMakowski, 2020-03-28:
Yeah, with a `silent=False` arg to allow for turning off the printing

JanCBrammer, 2020-03-28:
Or maybe `warnings` instead of `silent`? Otherwise it becomes a bit of a double-negative (those hurt my head) :D

DominiqueMakowski, 2020-03-28:
in the past, I tended to use `verbose`, but in fact I realise that usually what I actually want to do is to really silence a function 

[image]

I guess it's more "natural" and less stressful for the user than to play with something called "warnings" ^^

JanCBrammer, 2020-03-28:
Fair enough, warning implies danger :D Let's silence it then.

JanCBrammer, 2020-03-28:
Closing after this has been resolved in #168

closed 2020-03-28 by JanCBrammer