neuropsychology/NeuroKit #174: PPG: Resources and Ideas

DominiqueMakowski, 2020-03-29:
## Features

- [ ] Cleaning
- [ ] Peak detection / correction
- [ ] Rate
- [x] HRV: available in ecg, we'll just need to feed in the rate
- [ ] Delineation & Cardiac Phase
- [ ] ...

## Resources

DominiqueMakowski, 2020-03-29:
@LegrandNico once we added PPG processing, I expect systole and neurokit to have quite some overlap in terms of features and code, so we should somehow try to join forces 💪 

I was looking at the [dependencies](https://github.com/embodied-computation-group/systole/blob/master/requirements.txt) of systole, and they're actually quite similar to ours (aside from our `sklearn` and your `seaborn`), which gave me an idea.

I don't know what your plans are for the future of systole, but as for now it seems to incorporate two aspects, namely *interfacing* (with hardware and software - e.g., psychopy) and *processing*. One idea could be for systole to externalize some of its processing parts to NK (while potentially, keeping your API, but rely on NK underneath). This would have the benefits of diminishing the number of redundant code/features and would be easier to maintain (the more contributors and users there are, the more it's easier to test-out, validate and improve). Naturally, if we were to join forces, systole's authors would be included, and cited, in the paper we are preparing ☺️

LegrandNico, 2020-03-30:
@DominiqueMakowski I agree that the ecosystem of physio data processing with Python would greatly benefit from the aggregation of all the many pieces into one general package, and I am happy to help NeuroKit be this one. 

Concerning Systole, we are going, for now, to pursue the development separately as this is something we will use in the lab and cite in the method section of our publications. I am a bit cautious about internally calling a package that is under active development as we may lose some control on what is happening as commits come in and it may add work for us with version compatibility. What I can do is to commit in parallel on Systole and Neurokit when I add function, if there are some you find interesting, and help you to develop some aspect. That way all the functionalities of Systole can be embedded and further developed inside NeuroKit. When NeuroKit is 100% tested and stable I am happy to use it and cite it, as you mentioned.

Currently, the peak detection algorithm in Systole is rather simple and borrow bits from many packages but has been really efficient until then for us. I am also working on a better implementation of the RR artefact detection algorithm from Lipponen & Tarvainen, which can also be used with ECG or other recordings of cardiac activity and might be of interest for NeuroKit.

DominiqueMakowski, 2020-03-30:
Fair points, you're right in that it may be too early to think of a hard-integration, let's see where the future brings us! 

> What I can do is to commit in parallel on Systole and Neurokit when I add function, if there are some you find interesting, and help you to develop some aspect

That sounds great, we can brainstorm some ideas and get your opinion once we have here a first working version of support. Currently, @JanCBrammer is our PPG (and ECG 😁) master (also the author of the great [biopeaks](https://github.com/JanCBrammer/biopeaks), and has started implementing the Elgendi M, Norton I, Brearley M, Abbott D, Schuurmans D (2013) method for peak extraction. We'll see how it goes!

JanCBrammer, 2020-03-31:
>  I am also working on a better implementation of the RR artefact detection algorithm from Lipponen & Tarvainen, which can also be used with ECG or other recordings of cardiac activity and might be of interest for NeuroKit.

@LegrandNico, I would really love for someone who also read the Lipponen & Tarvainen artefact detection paper to review [our implementation](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/ecg/ecg_fixpeaks.py). The paper isn't 100% clear at times (at least to me) :)

LegrandNico, 2020-03-31:
@JanCBrammer I didn't know that you implemented this one too :) That very timely as I just pushed a new version on the dev branch of Systole for [detection](https://github.com/embodied-computation-group/systole/blob/839a6dfbf2e112fe0b15b68f95d0d7f3e60408c0/systole/detection.py#L103) and [correction](https://github.com/embodied-computation-group/systole/blob/dev/systole/correction.py) of RR artefacts based on this algorithm. This is not fully tested for now and I'm not 100% sure it will find all the artefacts it should as the paper is a bit obscure sometimes for me too ^^. Importantly, however, here I dropped the `for` loop design I used in the previous version, which makes the code now run 10 times faster. It also returns more parameters and should fit more closely the paper.

If you want to work on this function we can open a collab notebook and see what is the best way to implement/test that.

JanCBrammer, 2020-03-31:
> If you want to work on this function we can open a collab notebook and see what is the best way to implement/test that.

@LegrandNico, absolutely let's do this. We can use artificial ECG or PPG signals to test the artifact detection and correction in controlled conditions.

> This is not fully tested for now and I'm not 100% sure it will find all the artefacts it should as the paper is a bit obscure sometimes for me too ^^

I mostly struggle with the _decision algorithm_ (section 2.1 in the paper). But I'm sure we'll get to the bottom of it :)

LegrandNico, 2020-04-01:
I created this Colab Notebook with the artefacts detection function and a rather simple RR time series simulation.

https://colab.research.google.com/drive/1sOItKglMQrKp4IR8anjem-3u4kQmhmBJ

We can co-author this function and the tutorial notebook for performance testing so we can add this version to our respective packages. @DominiqueMakowski this can be a nice tutorial for NeuroKit too.

Here the `rr_artefacts` function takes RR time-series as input and output all the relevant parameters from Lipponen & Tarvainen algorithm. The plots and corrections are performed by other functions. Another function could try to correct using the underlying ECG or PPG signal, eg. trying to find the closest peaks corresponding to the corrected RR values. This is a bit more complex and should come later, but I'm happy to work on it too.

What still needs to be done is from my perspective is:

- [ ] See if it actually works.

- [ ] Docstring: add a Notes section where the algorithm is fully explained (see example https://pingouin-stats.org/generated/pingouin.anova.html#pingouin.anova)

- [ ] Test for sensitivity and specificity. We can simulate RR time-series or use dataset, I am not super familiar with this.

- [ ] Test for execution time. Using pandas function all along the way might make this function even faster.

- [ ] Compare results with Kubios.

JanCBrammer, 2020-04-01:
@LegrandNico, thanks for taking initiative! I'm kind of swamped work-wise atm, will look into it asap (probably Wednesday next week). Do you have a Kubios Premium license? I think the correction isn't availabe in the Standard version.

DominiqueMakowski, 2020-04-02:
> Another function could try to correct using the underlying ECG or PPG signal, eg. trying to find the closest peaks corresponding to the corrected RR values.

Yes we also thought about back-informing the R peaks detection and location based on the corrected rate, but yeah that's tougher than it seems :/

https://github.com/neuropsychology/NeuroKit/blob/053ed688e153bce9f87c5a4a0a33364c7e7eb9f8/neurokit2/signal/signal_fixpeaks.py#L43-L56

[image]

*(original peaks in blue and reconstructed peaks in red)*

sangfrois, 2020-05-10:
I found a method that could be relatively easy to implement : https://www.ncbi.nlm.nih.gov/pubmed/29059822

I'll do my best to code it and will post my attempts here.

closed 2021-05-22 by stale[bot]