neuropsychology/NeuroKit #176: ecg_segment()

DominiqueMakowski, 2020-03-29:
Since we use it a lot, we can have an exported function to extract individual heartbeats. Todo:

- [ ] docstring
- [ ] Add a show arg to plot them
- [ ] Use it in the individual heartbeats extraction example
- [ ] Fix beat window:

```python
ecg = nk.ecg_simulate(duration=15, sampling_rate=1000, heart_rate=80)
signals, info = nk.ecg_process(ecg, sampling_rate=1000)
nk.ecg_plot(signals, sampling_rate=1000)
```
[image]

zen-juen, 2020-03-30:
The beat window's issue seemed to be with extracting the correct heart rate in the `ecg_segment` internal function where:
` heart_rate = np.mean(np.diff(rpeaks) / sampling_rate * 60) `
 Have changed it to:
` heart_rate = np.mean(ecg_rate(rpeaks, sampling_rate=sampling_rate))`

DominiqueMakowski, 2020-03-30:
Yup it fixed things 👏 

The new window is perfectly for slow BPMs: 

## 50 BPM:
[image]

## 120BPM

[image]

But for faster BPMs (> 80) we might wanna increase the window a tiny bit so that we have the offset and onsets of the extremities

zen-juen, 2020-03-30:
Added a constant to expand the scale for if higher rates like so:
```
    if heart_rate >= 80:
        c = 0.1
        epochs_start = epochs_start - c
        epochs_end = epochs_end + c
```

Looks better 😄 
[image]

DominiqueMakowski, 2020-03-30:
OR rather than a constant, you add a portion of the square of the scaling factor ^^ so that the more the rate increases the more it expands in a non-linear fashion 🤷‍♂ 

Although if that gets too complicated don't bother too much it's fine like this 😄

closed 2020-03-31 by zen-juen