neuropsychology/NeuroKit #180: Change interpolation method in rsp.rsp_rate()

Louis-Deschamps, 2020-03-31:
**Describe the solution you'd like**
At the end of rsp_rate(), an interpolation is done on the extracted respiration rate. The method for the interpolation is "quadratic". On very clean data, the quadratic method gives a smooth and nice respiration rate but in certain conditions (noisy data), this leads to extreme or even negative respiration rate values. I think it would be nice to be able to choose the interpolation method. 

**How could we do it?**
The addition of an argument like "interpolation_method" in the function rsp_rate() would allow the user to choose which interpolation method he wishes to use.
[image]

DominiqueMakowski, 2020-03-31:
Mmh interesting... 🤔  

one of the reasons why rate interpolation is often using splines (e.g. cubic splines) is because it is the closest to the underlying physiological changes. Let's say you have three points [0, 1, 0], if you connect them straight (linear interpolation) it makes a triangle, but physiologist *assume* that the physiological mechanism behind is, in fact, rather smooth, hence an interpolation that gives these U shapes. However, indeed sometimes it might extrapolate too much.

Now in your case what strikes me the most is the change rate, there seem to be very big and sudden changes (a respiration rate going virtually from 0 to 150, and then back in a few seconds), and also the baseline trend is quite low (close to 0). May I ask whether it is human data? The problem in your case might not be the interpolation method, but rather the peak detection that is incorrect. The rate depends on the peak intervals, itself depending on peak detection. Maybe you could share a plot with the raw signal and the identified peaks to see if that could be one of the reason?

Louis-Deschamps, 2020-03-31:
Hi Dominique. Thanks for your very quick answer. Yes, it is human data. You are right, the peak detection is clearly the first cause of the problem here. It is because the peak detection algorithm missed a lot of peaks that the quadratic interpolation failed that much. But peak detection will never be perfect on respiration signals, which a often very noisy. I think a quadratic interpolation will amplify the fails of the detection algorithm.

I understand your argument of trying to smooth the output to have a more "physiological" behavior. But in situations where the signal is quite noisy, the smoothness of the output is not a priority. Indeed, the respiration rate is very often averaged afterwards over a certain time window. In these conditions, I think a linear interpolation would allow a better estimation of the respiration rate, even if the plot would indeed look a bit "unnatural". That is why I think it could be interesting to have both options :)

JanCBrammer, 2020-03-31:
I also always use linear spline interpolation for the exact reasons @Louis-Deschamps describes (physiologically implausible values with cubic interpolation on noisy extrema). @Louis-Deschamps, could you show us a plot of the raw breathing data that cause this behavior?

DominiqueMakowski, 2020-03-31:
@JanCBrammer @Louis-Deschamps aside from the peak-fixing issue, I see two ways of addressing the rate issue:

- We can add an `interpolation_method` switch to all the `*_rate()` functions, so that people can mitigate the variations.
- We could enable the imposition of a cap, - a tunnel: We'd compute the 99% bound of the rate variations (from 0 to max, without including negative values). That would return us let's say [6, 54]. Then we would cap all the values above and below these bounds, so they would remain "in the tunnel". Then, user could change this tunnel width, by setting it for instance to 95% or 90%. This has the advantage of only impacting potentially artifacted regions.

What do you think?

Louis-Deschamps, 2020-04-01:
I think both options are interesting :) I believe it is important to have control over which interpolation method is used so I would prefer option 1. But the second one could be a good thing to add too !
Here is a plot of the raw data, the clean data and the respiration rate:
[image]

Louis-Deschamps, 2020-04-01:
And here is a focus on the second half of this signal (from 300s to 600s), with the peaks extracted with khodadad method.
[image]

JanCBrammer, 2020-04-01:
@Louis-Deschamps , thank you for the plots!

https://github.com/neuropsychology/NeuroKit/blob/e53312c049ef262ce18264c475c0a1e3569f5468/neurokit2/rsp/rsp_findpeaks.py#L158-L164

I think the criterion above might cause the large number of non-detected (false negative peaks). Since `min_diff` is based on the mean off all peak-to-through differences, the criterion becomes very conservative / strict when there are large outliers in peak-to-trough differences. An example would be the outlier between 5000 and 6000 samples in the plot above. 

@DominiqueMakowski. we could base the criterion on the median instead. Or maybe on the 80th quantile?

>  I believe it is important to have control over which interpolation method is used so I would prefer option 1.

+1

DominiqueMakowski, 2020-04-01:
@Louis-Deschamps That signal has seriously artifacted portions indeed 🙉 

Maybe you could try with a stronger cleaning (in particular a much stronger filter)?

> @DominiqueMakowski. we could base the criterion on the median instead. Or maybe on the 80th quantile?

Median sounds good, as we can assume that in clean cases it will be close to mean, but less biased in artifacted cases.

> expose interpolation method

Should we name the argument `method_interpolation` or `interpolation_method` 🤔

JanCBrammer, 2020-04-01:
> Should we name the argument `method_interpolation` or `interpolation_method`

`interpolation_method` sound good to me.

JanCBrammer, 2020-04-01:
@DominiqueMakowski, pushed two commits to open PR on dev.

Louis-Deschamps, 2020-04-01:
> @Louis-Deschamps That signal has seriously artifacted portions indeed 🙉

Yes it does ! But it's not the worse signal I have... Respiration signals can gen very dirty when the subject is moving and/or talking...

> Maybe you could try with a stronger cleaning (in particular a much stronger filter)?

Yes, it usually helps a bit :)

DominiqueMakowski, 2020-04-01:
@Louis-Deschamps can you try with the current dev branch and see if it improves?

Louis-Deschamps, 2020-04-01:
As my signal is very noisy it's still pretty bad. But with a bit more pre and post processing, I should manage to get something out of it ! Having the linear interpolation helps me see exactly where the problems are. Thanks a lot @DominiqueMakowski  and @JanCBrammer  for being so reactive ! And thanks for developing Neurokit, it's a very good package :)

closed 2020-04-01 by DominiqueMakowski