neuropsychology/NeuroKit #186: Optimal Embedding Dimension and Time Delay

DominiqueMakowski, 2020-04-05:
Alright, so last night I stumbled across this [document](https://www.biopac.com/wp-content/uploads/Specialized_Analysis_Acq39.pdf) from Biopac's Acqknowledge (their processing software), in which they describe some of their features. Page 340, they talk about their "Chaos" module to compute complexity related stuff. Two things caught my attention.

- **Optimal Embedding Dimension**: Most complexity functions have an embedding dimension argument (order 'm'), usually set to 2. However, it seems to be possible to get the optimal 'm' by finding the smallest values where the correlation dimension reaches a local maximum. Would it make sense to support that?

- **Optimal Time Delay**: In order to plot the chaotic attractor (something that would also be nice to implement) and to find the optimal embedding dimension, they have a function that computes the optimal time delay. I'm not even sure it makes some sense.
  - Based on [this post](http://node99.org/tutorials/ar/), I added [this function](https://github.com/neuropsychology/NeuroKit/blob/dev/neurokit2/complexity/optimal_delay.py), but I am very unsure of its implementation, as the original isn't reproducible. 
  - I was wondering if this "time delay" Tau is the delay used in [`_embed()`](https://github.com/neuropsychology/NeuroKit/blob/dev/neurokit2/complexity/utils_embed.py#L5) (used in most of the complexity functions), where it is set to 1. If so, wouldn't it make sense to use the optimal Tau, rather than 1?

@CSchoel do you have any thoughts?

CSchoel, 2020-04-08:
I actually thought about this topic a lot. On the one hand, it is crucial - especially for novices - to have a sound guideline for choosing parameters. Whenever I found recommendations in a paper I tried to implement them or at least reference them in nolds. Without guidelines, there is always the danger that you can adjust the parameters arbitrarily to basically get any result that you want - or that your findings may be wrong, because you used a standard setting that is not appropriate for your case.

On the other hand, my experience so far tells me that there usually is no such thing as a one-size-fits-all-solution for most parameters. Automatic procedures perform well with a narrow range of data - usually they work well for data that *is* chaotic and fail for data that is not. This alone is problematic, if you have an experiment that wants to find out whether there is chaos in your data and the result could go either way. This is the reason why all my nolds measures have a debug_plot parameter. I would always suggest to have a look at what the algorithm is actually fitting before drawing any conclusions about its output. My feeling is that finding the right setting of parameters can be more art than hard science and in any case requires an intimate knowledge of the algorithm as well as the data that is being analyzed.

Long story short: I believe it would be worthwhile to implement algorithms for choosing parameters and to use them as default choice, but they should always be taken with a grain of salt and the documentation should make that transparent.

Regarding the time delay: I will need to have a more detailed look at the suggested function, but I think you are right that the delay in _embed and tau from the post are the same parameter. I would refrain from using a delay != 1 in algorithms that do not list delay as a parameter that can be varied, but for some algorithms it may be interesting to choose the delay based on an automatic procedure. I think I remember that there is something like this for one of the algorithms in nolds. As always I cannot guarantee when I will have the time, but I will try to investigate. :male_detective: :wink:

DominiqueMakowski, 2020-04-09:
I think it'd be good to provide such an option and at least try to implement such a tool (could be useful for testing, exploration or validation). And indeed, if we were to include it as a default parameter, it'd have to be thoroughly explained! 

Another side thing that bugs me is that in all of the above, if I understand correctly, the parameters are expressed in terms of samples. But in physiology usually, the sampling rate is more relevant; you can have two identical signals, sampled at different rates. So would it make sense to adjust for the sampling rate of the input signal, either by resampling it to a fixed SR, or by adjusting the parameters (tau, m, ...?) as a function of sampling rate?

DominiqueMakowski, 2020-05-02:
I fell back into this, and started implementing several methods to estimate the optimal delay, with cool visualization helpers:

```python
# Simulate heart rate signal
ecg = nk.ecg_simulate(duration=60*6, sampling_rate=150)
signal = nk.ecg_rate(nk.ecg_peaks(ecg, sampling_rate=150)[0], sampling_rate=150)

# Visualize optimal delay
tau = nk.embedding_delay(signal, delay_max=1000, show=True)
```
[image]

```python
# Visualize reconstructed attractor
embedded = nk.embedding(signal, delay=182, dimension=3, show=True)
```
[image]

We will now into some tools for the estimation of the dimension m. Feel free to give us some feedback ☺️

closed 2020-05-13 by Tam-Pham