neuropsychology/NeuroKit #191: Include quality in ECG plot

DominiqueMakowski, 2020-04-09:
We could include the ECG signal quality in the raw/cleaned signal plot, as a background green/blue area. The only thing prior to plotting it is to rescale (`nk.rescale()`) the quality to the max and min of the signals (because they define the scale of the plot).

zen-juen, 2020-04-09:
This is defined to the max and min of the cleaned ECG signal 

[image]

DominiqueMakowski, 2020-04-09:
how does it look as an area (similar to pandas.area.plot()) from the minimum value to the y? and behind the plot?

zen-juen, 2020-04-09:
You mean like this?

[image]

DominiqueMakowski, 2020-04-09:
Yeah, I'd even remove the blue outline and increase the alpha to make it even more delicate 😬

zen-juen, 2020-04-10:
[image]

DominiqueMakowski, 2020-04-10:
noice! need to fix the legend tho so it's not transparent. And maybe let's try with green #4CAF50 😏

closed 2020-04-10 by DominiqueMakowski