neuropsychology/NeuroKit #198: `nk.eda_plot` doesn't work for sampling rates other than 250hz

dschreij, 2020-04-26:
**Describe the bug**
A description of what the bug is.
`nk.eda_plot` throws an IndexError when the sampling rate of the dataset is not 250Hz

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
```python
s, i = nk.eda_process(eda_signal, sampling_rate=250)
nk.eda_plot(s) #works
s2, i2 = nk.eda_process(eda_signal, sampling_rate=10)
nk.eda_plot(s2, sampling_rate=10) #doesn't work
```
This throws the error with stacktrace:
```

nk.eda_plot(signals.reset_index(drop=True))
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
analysis.py in 
----> 1 nk.eda_plot(signals.reset_index(drop=True))

~\Miniconda3\envs\main\lib\site-packages\neurokit2\eda\eda_plot.py in eda_plot(eda_signals, sampling_rate)
     61 
     62     # Mark segments.
---> 63     risetime_coord, amplitude_coord, halfr_coord = _eda_plot_dashedsegments(eda_signals, ax1, x_axis, onsets, peaks, half_recovery)
     64     risetime = matplotlib.collections.LineCollection(risetime_coord, colors='#FFA726', linewidths=1, linestyle='dashed')
     65     ax1.add_collection(risetime)

~\Miniconda3\envs\main\lib\site-packages\neurokit2\eda\eda_plot.py in _eda_plot_dashedsegments(eda_signals, ax, x_axis, onsets, peaks, half_recovery)
    102     # Half recovery.
    103     halfr_end = scat_halfr.get_offsets()
--> 104     halfr_start = [(peak_top[i, 0], halfr_end[i, 1]) for i in position]
    105     halfr_coord = [(halfr_start[i], halfr_end[i]) for i in position]
    106 

~\Miniconda3\envs\main\lib\site-packages\neurokit2\eda\eda_plot.py in (.0)
    102     # Half recovery.
    103     halfr_end = scat_halfr.get_offsets()
--> 104     halfr_start = [(peak_top[i, 0], halfr_end[i, 1]) for i in position]
    105     halfr_coord = [(halfr_start[i], halfr_end[i]) for i in position]
    106 

~\Miniconda3\envs\main\lib\site-packages\numpy\ma\core.py in __getitem__(self, indx)
   3186         # mask of being reshaped if it hasn't been set up properly yet
   3187         # So it's easier to stick to the current version
-> 3188         dout = self.data[indx]
   3189         _mask = self._mask
   3190 

IndexError: index 2 is out of bounds for axis 0 with size 2
```

**Expected behaviour**
A clear and concise description of what you expected to happen.

`nk.eda_plot` showing the figures as it shows for a set with a sampling rate of 250, as displayed at the bottom of: https://neurokit2.readthedocs.io/en/latest/examples/eda.html

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Specifiations (please complete the following information):**
 - Package Version [e.g. 0.2.1]
v0.0.18

dschreij, 2020-04-26:
This also still occurs with v0.0.27 that I got from master.

zen-juen, 2020-04-28:
Hi there @dschreij! It seems like the problem lies within some internals that `eda_process` relies on, as it doesn't accurately detect the number of half-recovery time points for `sampling_rate=10`. Thanks for raising this up with us, we'll fix it and get back to you asap :)

dschreij, 2020-04-29:
Great! Thanks. I already wrote my own plotting function in  the meantime, but I'm sure others will be happy it's fixed.

DominiqueMakowski, 2020-04-29:
Feel free to share if you have some ideas or suggestions :)

dschreij, 2020-05-01:
Sure! here we go.

## Retain original index

One thing that I would to suggest is to honor the index type of the original dataframe (or rather Series) that you process with `nk.eda_process()`. In my case this index was of pandas.Timedelta but it was lost in the result of the function (which just had an incrementing index again). For Timedelta the index was easy enough to recreate by doing:

```python
    signals, _ = nk.eda_process(df['SCL'], sampling_rate=10)
    # Convert index to timedelta. Since the sampling rate is 10Hz, and the index just increments
    # by 1, multiply its values by 100 and convert with ms as the unit
    signals.index = pd.to_timedelta(signals.index*(1000/sampling_rate), unit='ms')
```
but for Timestamps, the index has to be obtained again from the source dataframe, with for example:
```
signals, _ = nk.eda_process(df['SCL'], sampling_rate=10)
signals.index = df.index
```
if you use a time-based index, the signal gets nicely plotted with the time (delta) shown on the x-axis, instead of the sample no. which was at least preferably for my specific use case. So it would be nice if some mechanism would be in place in the function, which you can specify (by a parameter?) to recreate or retain the original index of the source Series/Dataframe.

## Plotting function

My plotting functions were quite simple so I can place them here if you would like to see what I did (even though you already have great stuff in place, so disregard it if your fix works). It's all annotated so you need python >3.5 for it to work

```python
from typing import List

import matplotlib.pyplot as plt
import numpy as np
import neurokit2 as nk
import pandas as pd
import seaborn as sns

sns.set()

def addAxisLabels(axes: np.ndarray, titles: List[str]) -> None:
    ''' Add titles and axis labels to each subplot '''

    for ax, title in zip(axes, titles):
        ax.set_ylabel('EDA (uS)')
        ax.set_title(title)
        # only add x-axis label to bottom subplot.
        if ax == axes[-1]:
            ax.set_xlabel('Time (H:M:S)')

def addPeakLines(df: pd.DataFrame, ax: plt.Axes) -> None:
    ''' Add lines at each SCR peak. Draw from the bottom of the figure,
       up to the line depicting the SCL '''

    peak_times = df[df['SCR_Peaks'] > 0]
    bottom, _ = ax.get_ylim()
    for t, data in peak_times.iterrows():
        ax.plot(
            [t.value]*2,
            [bottom, data['EDA_Clean']],
            color='r',
            linewidth=1,
            zorder=0,
            linestyle='dotted'
        )

def plotSignals(signals: pd.DataFrame) -> None:
    translations = {
        'EDA_Clean': 'Cleaned signal',
        'EDA_Tonic': 'Tonic component',
        'EDA_Phasic': 'Phasic component'
    }

    with sns.axes_style('white'):
        axes: plt.Axes = signals.rename(columns=translations)[[
            translations['EDA_Clean'],
            translations['EDA_Tonic'],
            translations['EDA_Phasic']
        ]].plot.line(
            subplots=True, sharex=False, figsize=(11, 13), legend=None)
        addAxisLabels(axes, titles=list(translations.values()))
        addPeakLines(signals, axes[0])
        plt.tight_layout(rect=[0, 0.03, 1, 0.95])
        sns.despine()

```
I hope it is of some use to you!

dschreij, 2020-05-01:
And of course feel free to close the issue if you regard it solved!

commit cfad29ce (2020-04-28): fix

#198
commit d215af6f (2020-04-28): fixes

#198
closed 2020-05-06 by DominiqueMakowski