neuropsychology/NeuroKit #202: Improving the high-level data processing pipelines

hungpham2511, 2020-05-06:
**Describe the solution you'd like**
Hi guys,

Looking at the NeuroKit's tutorials the other day, it occurs to me that the high-level processing functions, i.e., `bio_process()' at 
https://github.com/neuropsychology/NeuroKit/blob/471a2e8dbe442bc3da4dbe417db787ec4b574236/neurokit2/bio/bio_process.py#L13 

 are somewhat restrictive. 

Suppose I have a signal to process, but my workflow is not exactly the same as `bio_process`, then I must "copy" the bio_process function to my own script and do those modifications. A disadvantage of this is I will lose other "fancy" processing that you guys will be adding to this high-level function in newer versions. Or those carefully tuned parameters in the QRS detection algorithm. 

That is assuming that I am already a power-user, that understands python quite well to be able to find this function in the source file. This is actually quite tricky if I install NeuroKit using pip! Further, the process function can be so large that I must try to understand everything before modifying it.

In fact, this is a problem even if the default processing steps are perfectly suitable. When I do my own analysis, I often need to preview the processing steps to make sure that everything is good and that I can trust the final result. Doing so would also require changing the source code (to add plot, etc, print values).

**How could we do it?**

The key point is to allow the user to re-define part of the code without having to access the source code. A possible solution is to use class inheritance. See the below pseudocode for some high-level idea.

```python
# library code
class BioProcess:
  def read_data(csv_file):
       # default implementation
  def process_ecg(self, df):
       # default implementaition
  def report_result(self):
       # default: show plot, print out stuffs ,etc...

# user code

class MyOwnBioProcess(BioProcess):

  def read_data(csv_file):
    # my own implementation and processing steps

pipeline = MyOwnBioProcess('test.cvs')
pipeline.run()

```

JanCBrammer, 2020-05-06:
> When I do my own analysis, I often need to preview the processing steps to make sure that everything is good and that I can trust the final result. Doing so would also require changing the source code (to add plot, etc, print values).

This is also my experience. 

> allow the user to re-define part of the code without having to access the source code. A possible solution is to use class inheritance.

+1

I think that this implementation is even more applicable in the modality-specific wrappers, for instance 
https://github.com/neuropsychology/NeuroKit/blob/8b4c62258cbcaa634327bfeeb032f0f06360ace8/neurokit2/ecg/ecg_process.py#L12

DominiqueMakowski, 2020-05-06:
I see what you mean, but would require rewriting the whole API to go (back #32 😬) to a class-based API. (maybe I'm exaggerating, cause in reality we could relatively easily add classes on top of the current architecture, but I'm not sure having mixed-API would be super intuitive 😕)

Before going there, is there a way to emulate this behaviour, or at least facilitate it achievement, with the current organization?

One thing that comes to my mind to facilitate this pipeline customizing is to use strings. For a start, we could have a function:

```python
import re
import inspect

def get_sourcecode(function):

    # Access source code
    source = inspect.getsource(function)

    # Remove docstrings
    source = re.sub(r'\"\"\"(?:.|\n)+\"\"\"', '', source)

    return source
```

A user could call this function to get the source, then he could copy and paste the output and modify the bits hat he wants:

```python
import neurokit2 as nk
print(get_sourcecode(nk.emg_process))
```
```
def emg_process(emg_signal, sampling_rate=1000):
    
    # Clean signal
    emg_cleaned = emg_clean(emg_signal, sampling_rate=sampling_rate)

    # Get amplitude
    amplitude = emg_amplitude(emg_cleaned)

    # Get onsets, offsets, and periods of activity
    activity_signal, info = emg_activation(amplitude,
                                           sampling_rate=sampling_rate,
                                           threshold='default')

    # Prepare output
    signals = pd.DataFrame({"EMG_Raw": emg_signal,
                            "EMG_Clean": emg_cleaned,
                            "EMG_Amplitude": amplitude})

    signals = pd.concat([signals, activity_signal], axis=1)

    return signals, info
```

It's clearly a hack tho, but 🤷 

If we were crazy, we could think of having identifiable chunks through comments in the source code, for instance `# STEP CLEANING` to then be able modify these steps in the source using another function, like `nk.emg_process_modify('cleaning', mynewcleaningfun)` 😁

hungpham2511, 2020-05-06:
> I see what you mean, but would require rewriting the whole API to go (back #32 ) to a class-based API. (maybe I'm exaggerating, cause in reality we could relatively easily add classes on top of the current architecture, but I'm not sure having mixed-API would be super intuitive )

I wouldn't say it is that big of a change. You can think of the architecture as having two halves: the core library stuff which is purely functional and the make-the-user-experience better which can employ classes. And since NeuroKit is in python, you can always make a class "quack" like a function, like so:

```python
class MyProcess:
    def __call__(self, param_1, *args, **kwargs):
        # do something

MyProcess()(param_1, *args, **kwargs)
# almost identical to bio_process(param_1, *args, **kwargs)
```

In fact, I can do:
```python
def bio_process(*args, **kwargs):
    return DefaultBioProcess()(*args, **kwargs)
```

Now power users can do whatever they want and novices can simply call `bio_process` as before.

> Before going there, is there a way to emulate this behaviour, or at least facilitate it achievement, with the current organization?
>
> One thing that comes to my mind to facilitate this pipeline [...]

This is certainly possible. But trust me, doing source code parsing is not easy at all... You are basically building a preprocessor system, which is very very hard to do well.

JanCBrammer, 2020-05-06:
@DominiqueMakowski, parsing the source code doesn't really adress the copy-paste problem. It only makes copy-pasting more convenient. As @hungpham2511 already mentioned, on the downside that slight increase in convenience for the user requires a lot of work on the part of the developers (ensuring that the source code is properly annotated etc.). In the end the amount of work we have to put in would probably be equal to or even more than trying an object-oriented solution (which after all is where Python shines).

JanCBrammer, 2020-05-06:
@DominiqueMakowski, @hungpham2511, maybe we could draw some inspiration from scikit-learn's [implementation of pipelines](https://github.com/scikit-learn/scikit-learn/blob/master/sklearn/pipeline.py): 

+ https://arxiv.org/pdf/1309.0238.pdf
+ https://towardsdatascience.com/scikit-learn-design-principles-d1371958059b

closed 2021-05-22 by stale[bot]