neuropsychology/NeuroKit #203: Optimize a line of code

DominiqueMakowski, 2020-05-08:
Performance masters, I call upon you 🧙 Part of the following code seems inefficient:

```python
import sklearn.neighbors

# Variables
embedded = np.array([np.arange(4), np.arange(1, 5)]).T
r = 0.2

# This computes the pairwise distance between all points (a square matrix)
dist = sklearn.neighbors.DistanceMetric.get_metric("chebyshev")
dist = dist.pairwise(embedded)

# This is where performance is lost IMO
sim = np.exp(-np.linalg.matrix_power(dist, 1) / r)
count = np.sum(sim, axis=0)
```

Because it applies a transformation to the whole matrix which doesn't scale up. Somehow I feel like like it could be improved as the distance matrix is sort of a "repetition".

I tried computing the transformation from only one row of the matrix (`np.exp(-np.power(dist[0, :], 1) / r)`), but then not sure what to do with it to retrieve the same `count` vector...

@hungpham2511

hungpham2511, 2020-05-08:
Hi Dom, I am definitely not a "performance master" :) 

At at first glance, I can't see any problem with this code though. Tried your code on my interpreter and it's done in a few usec:

```
In [4]: timeit sim = np.exp(-np.linalg.matrix_power(dist, 1) / r)
   ...:
4.54 µs ± 87.8 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)
```

DominiqueMakowski, 2020-05-08:
I made a small example, but in reality the distance matrix can be huge more than 10000*10000:

```python
embedded = np.array([np.arange(10000), np.arange(1, 10001)]).T
```

hungpham2511, 2020-05-08:
Can I know what is the mathematical equation that you are trying to compute?

What is the rationale behind raising matrix power to 1 ? Wouldn't it just give you the same matrix.

Bests,

Hung

DominiqueMakowski, 2020-05-08:
right, cause the power is actually an argument and can be more than 1 (although I'll change it so that if power == 1 it doesn't pass through np.power, although not sure if it'll changes anything)

DominiqueMakowski, 2020-05-08:
It's a formula of one of the component of fuzzy entropy 

https://github.com/neuropsychology/NeuroKit/blob/d9926f815728a4e7145b161dcaa2b59cd4acc127/neurokit2/complexity/utils.py#L102-L109

hungpham2511, 2020-05-08:
> Because it applies a transformation to the whole matrix which doesn't scale up. Somehow I feel like it could be improved as the distance matrix is sort of a "repetition".

From the algorithmic perspective, you are right to suspect that there is repetition. However, this repetition is not as high as you might think. Suppose you can remove all the repeating numbers, I think you still have to deal with N^2 / 2 coefficients. This is only a 50% improvement and it's generally not worth pursuing.

The main issue here is the cost of computing the exp and matrix_power. The first requires O(N^2) operations. The second requires ~ O(N^3) operations. Now, these are hard theoretical bounds that are unlikely to be overcome.

In another word, even if you turn the numpy-based routines into loops, you still need to perform such number of operations, and hence, require about the same computational cost regardless.

DominiqueMakowski, 2020-05-09:
I tried to apply the transform only to the upper triangular and then copy it over to the lower triangular, it seems to lead to performance gains only when the power > 1, for some unclear reason.

```python
import neurokit2 as nk
import sklearn.neighbors

def original(dist, r=0.2, n=1):

    sim = np.exp(-np.linalg.matrix_power(dist, n) / r)
    count = np.sum(sim, axis=0)

    return count

def new(dist, r=0.2, n=1):

    # Get indices of upper triangular
    triang = np.triu_indices_from(dist)
    if n > 1:
        dist[triang] = np.power(dist[triang], n)
    # Apply transform to upper triangular
    dist[triang] = np.exp(-dist[triang] / r)

    # Copy upper to lower
#    dist[np.tril_indices_from(dist)] = dist.T[np.tril_indices_from(dist)]
    dist = dist + dist.T - np.diag(np.diag(dist))

    count = np.sum(dist, axis=0)
    return count
```

test:
```python
signal = nk.signal_simulate()
embedded = nk.embedding(signal, dimension=2)

dist = sklearn.neighbors.DistanceMetric.get_metric("chebyshev")
dist = dist.pairwise(embedded)

%timeit original(dist, n=1)
%timeit new(dist, n=1)

%timeit original(dist, n=2)
%timeit new(dist, n=2)

%timeit original(dist, n=3)
%timeit new(dist, n=3)
```

```
# power 1
1.72 s ± 20.7 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
4.22 s ± 44.3 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)

# power 2
13.2 s ± 743 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
6.5 s ± 11.6 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)

# power 3
24.4 s ± 2.09 s per loop (mean ± std. dev. of 7 runs, 1 loop each)
6.56 s ± 58.6 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
```

I wonder why it is slower for power 1 🤔 as if the copy pasting of the upper was more intensive than the exponentiating.

hungpham2511, 2020-05-09:
> ```python
>         dist[triang] = np.power(dist[triang], n)
> ```

You made a mistake here. This is not equivalent to matrix_power.

DominiqueMakowski, 2020-05-09:
When selecting `dist[triang]` is pulls a vector I think

hungpham2511, 2020-05-09:
If I am not wrong, np.power gives the coefficient-wise power, and not the matrix power. That's why it's faster.

hungpham2511, 2020-05-09:
Like I mentioned in the other reply, matrix_power is the expensive one. 

> The main issue here is the cost of computing the exp and matrix_power. The first requires O(N^2) operations. The second requires ~ O(N^3) operations. Now, these are hard theoretical bounds that are unlikely to be overcome.

DominiqueMakowski, 2020-05-09:
Thanks, turns out I wanted the regular power! I replaced with this version:

```python
def new2(dist, r=0.2, n=1):

    if n > 1:
        sim = np.exp(-dist**n / r)
    else:
        sim = np.exp(-dist / r)
    count = np.sum(sim, axis=0)

    return count
```

Which seem slightly more efficient when n==1.

Still, though, it still feels weird to compute 2 times the same thing (as the dist matrix is made of two symmetric triangles)...

closed 2020-05-09 by DominiqueMakowski