neuropsychology/NeuroKit #205: Corrected Approximate Entropy (CApEn): implement simple equation

DominiqueMakowski, 2020-05-08:
in [Shi, B., Zhang, Y., Yuan, C., Wang, S., & Li, P. (2017). Entropy analysis of short-term heartbeat interval time series during regular walking. Entropy, 19(10), 568.](https://www.mdpi.com/1099-4300/19/10/568/htm) they start by defining ApEn:

[image]

Which we have implemented.

Essentially Φ0 - Φ1 is done here:

https://github.com/neuropsychology/NeuroKit/blob/d9926f815728a4e7145b161dcaa2b59cd4acc127/neurokit2/complexity/entropy_approximate.py#L53-L54

and the Φs are obtained as:

https://github.com/neuropsychology/NeuroKit/blob/d9926f815728a4e7145b161dcaa2b59cd4acc127/neurokit2/complexity/utils.py#L16-L33

-------

So far so good for AmpEn. But they go on showing the formula for the **corrected** approximate entropy (CApEn): 

[image]

I'm clueless as to how to transform the content of the`_phi()` function to match the CApEn formula. Any hints are welcomed.

Tam-Pham, 2020-05-09:
After reading both Shi (2017) and [Porta (2007)](https://journals.physiology.org/doi/pdf/10.1152/japplphysiol.00293.2007), one correction to the biases pertaining to self-matches (the vector itself is counted in the number of vectors that are within r distance from itself) is:  

Shi (2017):
> In addition, when Nm+1i(r)=1 or Nmi(r)=1 that implies the occurrence of self-match, the ratio Nm+1i(r)/Nmi(r) in cApEn formula should be substituted with 1/N−(m+1)τ.

Porta (2007):
> This correction can be simply obtained by substituting the ratio Ni(L,r)/Ni(L - 1,r) with 1/(N - L +1) when Ni(L -1 1,r)  1 or Ni(L,r)  1 [this correction tends to adjust even the situation in which a small, and likely underestimated, given the shortness of series, Ni(L - 1,r), will produce an Ni(L,r)  1]

The solution proposed is to substitute the ratio Nm+1i(r)/Nmi(r) with 1/N−(m+1)τ  when Nmi(r) or Nm+1(r)=1, as when it happens, it means that only that vector is close to itself. Even though I don't get how the substitution is derived, since it's mentioned in both papers, it seems valid 😅 and I can kind of understand the rationale for this solution. 

We can implement this simply by having: the below chunk in _phi() 
```
correction = 1 / (len(signal) - (dimension + 1) * delay)
count1 = np.where(count1.astype(int) == 1, correction, count1)
count2 = np.where(count2.astype(int) == 1, correction, count2)
```

However, what I don't get it is another suggestion that is in Shi (2017) but not Porta (2007) is 

> In order to reduce the bias, a corrected ApEn (cApEn) algorithm has been proposed [31]. Briefly, if limiting the number of vectors to N−(m+1)τ for dimension m, Equation (2) can be rewritten as −1N−(m+1)τ∑N−(m+1)τi=1lnNm+1i(r)Nmi(r), 

While I understand the mechanism (reducing the number of vector by 1), I don't know how does that helps to address the self-bias problem. Couldn't find this proposed solution in Porta (2007) too even though Shi (2017) quoted Porta for cApEn 😅 

I'm unsure about implementing this second correction till we know what it is. What do you think?

DominiqueMakowski, 2020-05-10:
From my read the "other" proposition of Shi 2017 was merely a rephrasing of the first one? 🤷 

I'd say it's safe to follow the original proposition by Porta. 🎉

Tam-Pham, 2020-05-10:
Actually, after taking closer look at both papers, my interpretation above is over-simplifying the correction. 
The formulation of approximate ApEn given in both papers transformed the original phi equation so that the effect of the self-matches in the calculation of ApEn can be managed. Therefore, I decided to implement the corrected ApEn in entropy_approximate() function instead of changing the _phi() function.

closed 2020-05-13 by Tam-Pham