neuropsychology/NeuroKit #206: Detrended Fluctuation Analysis (DFA): Discrepancies with the dfa implementation

DominiqueMakowski, 2020-05-09:
In my quest for complexity (see #40), I came across this other great implementation of DFA (https://github.com/dokato/dfa) by @dokato

Currently, the two implementations (ours, based on nolds, and *dfa*'s one) are (almost) identical given some argument tuning:

```python
import neurokit2 as nk  # The dev branch
signal = nk.signal_simulate(duration=1)

# NeuroKit
windows = [ 32,  38,  45,  53,  64,  76,  90, 107, 128, 152, 181, 215, 256, 304, 362, 430]
nk.fractal_dfa(signal, windows=windows, overlap=False, rms=True)
> 1.9172948230960725

# dfa
dfa(signal)[2]
> 1.9172948230960705
```

How there are a few things that differ that I'd like to have some insights about (for instance from the respective experts @dokato and @CSchoel ☺️ ).

## log or log2

The cause of the very microscopic difference is that *dfa* uses `np.log2()` where ours uses np.log(). 

https://github.com/dokato/dfa/blob/5bd7e9d6acb278c17d2518d69d2ce5f7221860d4/dfa.py#L78

Although the difference is very small, it begs the question of which one is the "correct" one?

## RMS vs. ?

The other difference is that *dfa* explicitly computes the RMS (seems to be based on [Ihlen, 2012](https://www.frontiersin.org/articles/10.3389/fphys.2012.00141/full)). The difference between our (and thus nolds-based) implementation is illustrated here:

https://github.com/neuropsychology/NeuroKit/blob/460b61637c7db1417c760e55f14a9dfc6be7f1e1/neurokit2/complexity/fractal_dfa.py#L144-L152

I don't really understand the whys and hows of that though, so I wonder if there's a reason to favour one or the other.

## default rule for windows

By default, *dfa* uses the following rule to generate the sequence of windows size:

```python
scale_lim=[5,9]
scale_dens=0.25

scales = (2**np.arange(scale_lim[0], scale_lim[1], scale_dens)).astype(np.int)
scales
> array([ 32,  38,  45,  53,  64,  76,  90, 107, 128, 152, 181, 215, 256,
       304, 362, 430])
```

The docstring mentions that:

> scale_lim are boundaries of the scale, where scale means windows among which RMS is calculated. Numbers from list are exponents of 2 to the power of X, eg. [5,9] is in fact [2**5, 2**9]. You can think of it that if your signal is sampled with F_s = 128 Hz, then the lowest considered scale would be 2**5/128 = 32/128 = 0.25, so 250 ms.

By comparison, nolds' rule of thumb is:

```python
nk.range_log(4, 0.1 * len(signal), 1.2)
> array([ 4,  5,  6,  8,  9, 11, 14, 17, 20, 24, 29, 35, 42, 51, 61, 73, 88])
```

It seems like the windows obtained by default are much smaller. @CSchoel in nolds says:

> nvals should be equally spaced on a logarithmic scale so that each window scale hase the same weight. min(nvals) < 4 does not make much sense as fitting a polynomial (even if it is only of order 1) to 3 or less data points is very prone. max(nvals) > len(data) / 10 does not make much sense as we will then have less than 10 windows to calculate the average fluctuation

I wonder what is the rationale behind dfa's choice.

Thanks for your input!

Might also interest @raphaelvallat (+ an occasion to initiate contact 😁)

DominiqueMakowski, 2020-05-09:
Regarding the last point about the **default windows sequence**: [Ihlen (2012)](https://www.frontiersin.org/articles/10.3389/fphys.2012.00141/full) says the following (although in the context of multifractal DFA but I suppose it's applicable for DFA):

- **minimum window**

> The minimum segment sample size should be large enough to prevent error in the computation of local fluctuation RMS. The minimum segment size larger than 10 samples is a “rule of tumb” for the computation of RMS. Furthermore, the minimum sample size must be considerably larger than the polynomial order m to prevent overfitting of polynomial trend (see “Local Fluctuations Close to Zero?” above). Thus, minimum segment size of 10 samples might be too small for large trend order m (Kantelhardt et al., 2002). 

If I understand correctly:

```
window_min = 10 * order (by default, 1)
```

- **max window**

> In MFDFA1, the maximum segment size should be small enough to provide a sufficient number of segments in the computation of Fq in Matlab code 8. A maximum segment size below 1/10 of the sample size of the time series will provide at least 10 segments in the computation of Fq in Matlab code 8. 

If I understand correctly, for a signal of length *n*:

```
window_max = int(n / 10)
```

- **spacing**

> Furthermore, it’s favorable to have a equal spacing between scales when they are represented in plot(log2(scale),log2(Fq)) to obtain a optimal performance of the linear regression that estimates q-order Hurst exponent Hq.

If I understand correctly, the spacing should be inverse to log2.

So I apply this to a signal of size *n*, I can find the windows sequence with a number of windows *w* as follows:

```
n = 1000  # length of signal
w = 6  # desired number of windows

window_min = 10
window_max = np.int(n / 10)

windows = np.exp2(np.linspace(np.log2(window_min), np.log2(window_max+1), w, endpoint=True)).astype(np.int)
windows = np.unique(windows)  # keep only unique
windows
> array([ 10,  15,  25,  40,  63, 100])

# Spacing on the log 2 scale (a bit different because of the int approx)
np.diff(np.log2(windows)))
> array([0.5849625 , 0.73696559, 0.67807191, 0.65535183, 0.66657627])
```

Does that seem sound? 

For neurokit's implementation, the user could provide either a custom windows list, or an int, which would be used as *w* (the desired number) in this formula.

raphaelvallat, 2020-05-09:
Bonjour @DominiqueMakowski !

Nice to e-meet you! I've been closely following both your Python and R packages -- really impressive work! I'll let people with more expertise than me reply on the issue of the optimal windows lengths, but here are some thoughts on the two other points.

1) **Log base**:

Only the intercept will be affected by the base, not the slope:

[image]

Since we're only returning the slope in first-order DFA, I think it doesn't make any difference which base you use. Most likely, the very small differences that you get are caused by floating-point imprecision.

2) **Root mean square**:

I'm actually very curious to hear what others have to say here, as it is really not clear from the DFA papers (and [Wikipedia](https://en.wikipedia.org/wiki/Detrended_fluctuation_analysis)) which of the two approaches is the correct one. I've tried some tests with the below code, and overall the two methods seem to return very similar results

```
# Calculate root mean squares of walks in d around trend
rms = np.sqrt(np.mean((d - trend)** 2, axis=1))
# Method 1: Arithmetic mean across all subsequences
fluctuations[i_n] = np.mean(rms)
# Method 2: RMS across all subsequences (= quadratic mean)
fluctuations[i_n] = np.sqrt(np.mean(rms**2))
```

I believe one of the advantages of the RMS is that it also [takes into account](https://en.wikipedia.org/wiki/Root_mean_square#Relationship_to_other_statistics) the variance around the mean, but from a purely theoretical perspective, I don't know which one is better to reliably estimate DFA.

Hope this all makes sense!
Thanks
Raphael

dokato, 2020-05-12:
Hey @DominiqueMakowski , thanks for mentioning my implementation. I'm not sure if there's anything for me to add, I tried to stick as close as possible to Ihlen (2012), so in general their aguments apply here. Indeed log2, log10 shouldn't affect a slope, and interestingly I did exactly same tests as @raphaelvallat with RMS order - same conclusions, doesn't matter much.

BTW recently I started playing with DCCA as well which is multimodal extension of DFA, might be of interests.

Great job with `NeuroKit`! I'll take a closer look into it.

CSchoel, 2020-05-12:
Ah, this is really interesting!

**log or log2**

As @raphaelvallat said, it doesn't matter for the fitted slope. I was also confused at first since many papers do not state which base is used for the log.

**RMS vs. ?**

I had a quick look into the original Paper from Peng et al. - and in fact it turns out that there are *two* original papers: one from 1994 with the unrevealing title "Mosaic organization of DNA nucleotides" where DFA was first described and one from 1995 titled "Quantification of scaling exponents and crossover phenomena in nonstationary heartbeat time series" where DFA is more central to the focus and title of the paper. The 1994 paper is a little bit more vague in the algorithm description and calculates the variance instead of the standard deviation, so I think we should use the 1995 paper as a reference. This paper sums the square differences between the data and the local trend across *all* data points of the input data (since it uses non-overlapping windows), divides it by the number of data points and then takes the square root of the result.

This is actually not consistent with *either* of the implementations we are discussing here since both take the square root too early. I think what we should do is more like this:

```
flucs = [np.sum((w - trend(w))**2) / len(w) for w in windows]
f_n = np.sqrt(np.sum(flucs) / len(windows))
```

Looking at the implementations that I used as reference for nolds this seems correct:

- The [PhysioNet implementation](https://www.physionet.org/physiotools/dfa/dfa.c) states for the helper function `polyfit` that "the return value is the sum of the squared errors (chisq) between the (x,y) pairs and the fitted polynomial". These values are then divided by the number of windows times the window size in the main function `dfa`.
- The [R package fractal](https://cran.r-project.org/web/packages/fractal/index.html) returns `(sum(fit$residuals^2))` from their helper function `regression.poly`. These values are again divided by the number of windows times the window size and then the square root is taken.
- Unfortunately the [third reference implementation](http://bsp.brain.riken.jp/~juricap/mdfa/mdfaintro.html) is no longer available.

Interestingly, PhysioNet seems to follow the 1994 definition using the variance (I could not find any reference to a sqrt-function in the code) while the fractal package uses the more common definition from 1995. I am not familiar with Ihlen 2012 and out of time for today, but it seems to me that at least for nolds and NeuroKit2 something along the lines of the above code example should be the correct choice if we refer directly to Peng et al. 1995, which seems to be the most common reference.

DominiqueMakowski, 2020-05-13:
@CSchoel following your updated formula, it gives exactly the same results as the RMS method 🎉 

Another mystery solved ☺️ 

I tried 3 implementations, and the second one seems to be the more efficient (slightly, but it's always something 🙆‍♂️):

```python
    def f1(detrended):
        fluctuation = np.sqrt(np.mean(detrended**2, axis=1))
        fluctuation = np.sqrt(np.mean(fluctuation**2))
        return fluctuation

    def f2(detrended):

        fluctuation = np.sum(detrended**2, axis=1) / detrended.shape[1]
        fluctuation = np.sqrt(np.sum(fluctuation) / len(fluctuation))
        return fluctuation

    def f3(detrended):
        fluctuation = np.mean(detrended**2, axis=1)
        fluctuation = np.sqrt(np.mean(fluctuation))
        return fluctuation

    detrended = segments - trends

    %timeit f1(detrended)
    %timeit f2(detrended)
    %timeit f3(detrended)
```
```
25.6 µs ± 1.96 µs per loop (mean ± std. dev. of 7 runs, 10000 loops each)
18.5 µs ± 1.38 µs per loop (mean ± std. dev. of 7 runs, 10000 loops each)
26 µs ± 38 ns per loop (mean ± std. dev. of 7 runs, 10000 loops each)
```

Will update NK accordingly, thanks guys for your input ❤️ 

> BTW recently I started playing with DCCA as well which is multimodal extension of DFA, might be of interests.

@dokato Definitely! do keep us updated of your progress on DCCA and more and do let us know if we can offer any help ☺️ 

on a side note, we have implemented an initial version of multifractal DFA:

https://github.com/neuropsychology/NeuroKit/blob/4f2ad7d9c61a79c0c31781e7145846c16203054c/neurokit2/complexity/fractal_dfa.py#L156-L159

Though we'd love to get some feedback as I'm always doubting my understanding of equations 😬

DominiqueMakowski, 2020-07-04:
- [ ] Note for myself (and @Tam-Pham @zen-juen): check suggested improvements https://github.com/LRydin/MFDFA/issues/1

LRydin, 2020-07-08:
Hey everyone! Nice to see so many interested people working on DFA and MFDFA. I've been working on my python scripts for this here [MFDFA](https://github.com/LRydin/MFDFA) so I'd be more than interested in seeing how you are all tackling many of the issues that arise :).

Does anyone know an existent version of DFA or MFDFA in `R` by the way?

Cheers!

DominiqueMakowski, 2020-07-09:
Hi @LRydin! Welcome aboard the fluctuating discussion boat ☺️ 

Regarding R, there are implementations of [DFA](https://github.com/constantino-garcia/nonlinearTseries/blob/master/R/DFA.R) and [MDFA](https://github.com/mlaib/MFDFA), which might be helpful

LRydin, 2020-11-02:
Hey there NeuroKitters,

I'm sorry for the long absence, I had in mind returning to NK in August... but life got in the way. I don't know what's the status of DFA in NK at this stage, I have been working on my [MFDFA](https://github.com/LRydin/MFDFA) and I would very much like to contribute at some stage to NK... if there is something still to be done :)

Could anyone give me a nudge on the status of DFA and RMS methods and alike?

Cheers,
Leo

DominiqueMakowski, 2020-11-02:
> I had in mind returning to NK in August... but life got in the way

Wait what? it's not the beginning of 2020 anymore?? What happened to this year?! 😵 

Regarding the status, nothing changed much since last time. The code for M/DFA is [in this file](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/complexity/fractal_dfa.py) if you want to take a look. It's likely that you could indeed bring some of the improvements from MFDA to here. Also, as your package is on pypi, we could add some [tests](https://github.com/neuropsychology/NeuroKit/blob/master/tests/tests_complexity.py#L143) to test NK's implementation against it (if you're sure of it 😁).

Scope-wise, I think as we've mentioned somewhere else, the goal would be to have a function with good defaults that would return some unique index. And also a nice plot if asked for, currently it looks like this:

```python
import neurokit2 as nk

signal = nk.signal_simulate(duration=3, noise=0.05)
dfa1 = nk.fractal_dfa(signal, show=True)
```
[image]

```python
nk.fractal_mfdfa(signal, q=np.arange(-3, 4), order=2, show=True)
```
[image]

Maybe also some details could be put in the docstrings to help the users navigate in the parameters (and link to your package)

LRydin, 2021-01-21:
Hey everyone, hey @DominiqueMakowski,

I have forked and edited the [fractal_dfa.py](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/complexity/fractal_dfa.py) and I believe I have obtained some comvincing results. I have to adjust a few things, especially what you had above with changing the order of the fit. That was a bit of a misconception. I will do a PR in time, but I wanted to see what you have to say for now :)

Below is a simple example, which actually a "nasty" type of process called Lévy motion (or more simply, a set of random number extracted from an &alpha;-stable distribution)

### Nasty process, good results
#### First, just DFA:
```python
from neurokit2.complexity import fractal_dfa
from scipy.stats import levy_stable

# Generate 100000 points
alpha = 1.5
X = levy_stable.rvs(alpha=alpha, beta = 0, size=10000)

q=2

fractal_dfa(X, show=True, multifractal=False, q=q)
```
we get
[image]

#### working MFDFA:

```python
from neurokit2.complexity import fractal_dfa
from scipy.stats import levy_stable

# Generate 100000 points
alpha = 1.5
X = levy_stable.rvs(alpha=alpha, beta = 0, size=10000)

q=[2,3,5,10]

fractal_dfa(X, show=True, multifractal=True, q=q)
```
and we get
[image]

Important thing to note here are the theoretical results: for positive &alpha; the rule is simple &alpha;=1/q. Check the plots (the legents) and you see that the fits are good: `q=2` is `0.5`, `q = 3` is `0.333`, `q=5` is `0.2` and `q=10` is `0.1`.

The file I have so far edited on my fork is [here](https://github.com/LRydin/NeuroKit/blob/master/neurokit2/complexity/fractal_dfa.py). I will work on some tests using my MFDFA and nolds from @CSchoel .

DominiqueMakowski, 2021-01-22:
That looks awesome 😮! Feel free to PR whenever :) Is there a consensual way to "summarize" the different DFA values in MFDFA to get a single point-value? Or is this really not recommended?

LRydin, 2021-01-22:
:) Cool. I will PR when the tests are ready as well. For the MFDFA, there is something else called _singularity spectrum_, which takes in all these linear fits and turns them into a function of the dimention of the fractal and the "strenght of the multifractality". I will be adding this to my own MFDFA, so once that is done and working, I will port it here. With the singularity spectrum you can really have a "single measure" of multifractality, which is really something like &beta;≈0 means monofractal and &beta;>0 means multifractal.

DominiqueMakowski, 2021-01-22:
very interesting indeed, looking forward to seeing how it comes together. Let me know if ya need any help!

Tam-Pham, 2021-07-24:
Hi @LRydin, 

Just want to check in with you regarding your [fork](https://github.com/LRydin/NeuroKit/blob/master/neurokit2/complexity/fractal_dfa.py). 😄 
The changes that you made with the multifractal Hurst exponent are definitely very significant (and much-needed) corrections. 

Also, I recently looked at [Kantelhardt (2002)](https://arxiv.org/pdf/physics/0202070.pdf) and [Ihlen (2012)](https://www.frontiersin.org/articles/10.3389/fphys.2012.00141/full). It seems like the implementation for the *singularity spectrum* should be quite straightforward: 

```python
dfa = np.zeros(len(q))
for i in range(len(q)):
    dfa[i] = np.polyfit(np.log2(windows), np.log2(fluctuations[:,i]), 1)[0]

# from Eq. 13, to calculate q-order mass exponent:
tau_q = dfa * q - 1

# from E. 15, to calculate singularity exponent and singularity dimension:
sing_exp = np.diff(tau_q) / (q[1] - q[0])  #alpha, note here that q is evenly spaced
sing_dim = (q[: -1] * sing_exp) - tau_q[: -1] #F(alpha)
```

The plot of `sing_exp` vs `sing_dim` defines the spectrum. And a few indices of mutlifractality can be:
- `width_exponent` (max. sing_exp - min sing_exp)
- `height_dimension` (max. sing_dim - min. sing_dim)
- `mean_exponent`
- `mean_dimension`
- ...

Would you like to make a PR for your fork and we can perhaps work from there?

LRydin, 2021-07-24:
Hey @Tam-Pham! Yeah, I got lazy and never got around to finishing this. Actually, I already have the code ready for the singularity spectrum and other measures in the `dev-singularity` branch of my [MFDFA](https://github.com/LRydin/MFDFA/tree/dev) (see [here](https://github.com/LRydin/MFDFA/blob/dev-singularity/MFDFA/singspect.py)), so I can try to integrate/migrate the code over to NK2 :). 

I don't have the suggested `mean_exponent` and etc., but that is easy to include :). I'll try to get around to it, as well as include the same function I have the MFDFA code, since it uses a smart way to reshape numpy arrays that makes it rather fast :).

Hope you are all doing ok there :)!

Tam-Pham, 2021-07-26:
That's great @LRydin! We (think that we) are doing alright. Getting used to the new normal I guess. And trust me, "finishing things" is definitely not that easy for anyone in the past year 😆 😷 

Looking forward to the PR and do let us know if we can help in anyway ^^