neuropsychology/NeuroKit #208: Moving modality-unspecific functions to signal

JanCBrammer, 2020-05-11:
Some functions are duplicated across modalities. For example, `ecg_rate()`, `rsp_rate()`, and eventually `ppg_rate()`. All do the exact same. I suggest we rely on `signal_rate()` across modalities (e.g., calling `signal_rate()` in `*_process()`). The same for `ecg_fixpeaks()` and `ppg_fixpeaks()`.

Similarly, I suggest we introduce a separate `hrv` module, to avoid duplication like `ecg_hrv()` and `ppg_hrv()`.

By removing duplication we avoid "shotgun surgery" (having to modify code in muliple places if we want to\have to refactor) and make things less cluttered.

@DominiqueMakowski, I will restructure the repository accordingly if there are no objections (or good counter-arguments that I might not have thought of, which is entirely possible :grin:).

JanCBrammer, 2020-05-11:
See also #166

DominiqueMakowski, 2020-05-11:
I definitely agree about streamlining the code and avoiding duplications. 

> All do the exact same. I suggest we rely on signal_rate() across modalities (e.g., calling signal_rate() in *_process())

My only worry is that some users, let's say that work with ECG, might expect to find all the functions necessary for their pipeline under `ecg_*`. Having 4 `ecg_*` functions and then `signal_*` is a bit inconsistent (or maybe it's just me 😅)

One compromise would be to have `signal_rate()` do all the hard work, and then simply reexport as `ecg_rate = signal_rate`.  One can also use that technique to reexport with different defaults:

https://github.com/neuropsychology/NeuroKit/blob/0011265f6848383a9fbdc4247b2c999b1fec2728/neurokit2/complexity/__init__.py#L22-L26

The only objection that could be made is that it would create more confusion than clarity, but I don't think so, especially if we clarify the alias in the docs. For instance, for complexity:

https://github.com/neuropsychology/NeuroKit/blob/0011265f6848383a9fbdc4247b2c999b1fec2728/neurokit2/complexity/entropy_approximate.py#L14

So that the user will find `ecg_rate()` as they would expect, but also learn (if they want) that it's nothing more than an alias for `signal_rate()`.

--- 

now for hrv, yeah why not, we could indeed have a separate module with separate files for each "domain" (time-domain, freq domain etc.) and then one wrapper.

But what's with RRV in that case? Should it stay under RSP? or move to HRV 🤔

JanCBrammer, 2020-05-11:
> One compromise would be to have `signal_rate()` do all the hard work, and then simply reexport as `ecg_rate = signal_rate`

+1 for that approach. Basically redirecting the call to `signal_rate()` from modality-specifi aliases.

> now for hrv, yeah why not, we could indeed have a separate module with separate files for each "domain" (time-domain, freq domain etc.) and then one wrapper.

+1 for this also.

> But what's with RRV in that case? Should it stay under RSP? or move to HRV 🤔

I'll leave it in `rsp` for now (after all, HRV is for "heart" stuff) and we can still move it later if we feel that it belongs somewhere else.

DominiqueMakowski, 2020-05-16:
do you want me to start reorganizing that or maybe you know better, with the signal_rate() etc.?

JanCBrammer, 2020-05-16:
@DominiqueMakowski, already started on my fork. I'm done with the rate-related reorganization and about to finish up the HRV. Will make a PR soon.

DominiqueMakowski, 2020-05-17:
I think we made some small edits to some functions that might be impacted by the reorganization, I hope you're not gonna have too many conflicts 😓

JanCBrammer, 2020-05-21:
Changes have been merged in #212. Closing.

closed 2020-05-21 by JanCBrammer