neuropsychology/NeuroKit #214: Fixing examples

DominiqueMakowski, 2020-05-19:
We need to re-run all the notebooks and examples in the docs as some of them might be broken with the recent updates.

Also, note this comment: https://github.com/spatialaudio/nbsphinx/issues/447#issuecomment-630766064 about the `%matplotlib inline` that we can probably remove.

JanCBrammer, 2020-05-19:
Are we using [doctest](https://docs.python.org/3/library/doctest.html) already to check if the examples work?

It wouldn't hurt to also check all the arguments in the docstrings. I noticed that some include arguments that are no longer used, or that new arguments are missing.

DominiqueMakowski, 2020-05-20:
Yeah, initially doctest was supposed to run, but then... we need to look again into that

I wonder if there is any tool that checks if like all functions are documented, and that the documentation format is okay (does it have a title, a description, are all the parameters documented, do the examples run etc.). Something like R has 😬

JanCBrammer, 2020-05-21:
> I wonder if there is any tool that checks if like all functions are documented, and that the documentation format is okay (does it have a title, a description, are all the parameters documented, do the examples run etc.).

Weirdly enough there doesn't seem to be a tool that does all this. Or I wasn't looking hard enough. There are tools that check if docstrings are properly formatted, but not if they exist in the first place and if they contain all valid arguments.

DominiqueMakowski, 2020-05-21:
@Tam-Pham is currently trying to got `doctest` to run, and I'm watching from afar 'cause it looks quite hellish 😅

JanCBrammer, 2020-05-21:
Did some more digging. There seems to be a pylint extension called ["Parameter documentation checker"](https://docs.pylint.org/en/1.6.0/extensions.html) that does what we want.

Tam-Pham, 2020-05-21:
Hi all, 
After combing through the internet, I had added in a few lines of codes in our `travis.yml` and `pytest.ini` files. The changes can be found here ddceee3896df948f11b196f33f5be1bf32a472bf

I don't know if it's the best way but it is working. The code is basically combing through each folder to check each docstrings in our function files. 

Nevertheless, the integration time could be really slow if we count on travis to do the work (since for each time, it has to run through both the tests and the doctests). 

So Zen and I are building the doctests locally, by only keeping the doctest in `pytest.ini` 
```
[pytest]
addopts = --doctest-modules
doctest_optionflags= NORMALIZE_WHITESPACE IGNORE_EXCEPTION_DETAIL
```
and run through each folder one by one in our system. For instance:
```
C:\Users\Pham Thanh Tam\GitHub\NeuroKit>pytest --doctest-glob='*.py' neurokit2/complexity
```
This manual work should be done only for the first time running. In the future, we can count on travis to detect any errors.

zen-juen, 2020-05-21:
Following what @Tam-Pham said we'll probably put up a more concrete tutorial on how to run this for future contributors and also make a summary of docstring errors that you might expect to avoid 😄

JanCBrammer, 2020-05-21:
>  put up a more concrete tutorial

@zen-juen, that reminds me, do we mention somewhere already that we are using NumPy style docstrings?

DominiqueMakowski, 2020-05-21:
we should add that to the contribution guide indeed

DominiqueMakowski, 2020-05-21:
https://dan.yeaw.me/posts/github-actions-automate-your-python-development-workflow/ useful link for future reference

DominiqueMakowski, 2020-05-31:
- [ ] Improve testing
- [ ] Automated releases with automated NEWS log
- [ ] Automated check, and update of README figures
- [x] Check twine testpypi *before* publishing
- [ ] Automated linting
  - [ ] isort (https://github.com/olance/isort-action)
  - [ ] pylint
  - [ ] black
  - [ ] pycodestyle 
  - [ ] mypy
- [ ] Example gallery (#115)
- [ ] Plots in the sphinx API

DominiqueMakowski, 2020-06-01:
This is just flake8, pylint is like even worse ^^

```
neurokit2\bio\bio_analyze.py:12:1: C901 'bio_analyze' is too complex (19)
neurokit2\bio\bio_analyze.py:56:128: E501 line too long (129 > 127 characters)
neurokit2\bio\bio_analyze.py:59:128: E501 line too long (141 > 127 characters)
neurokit2\bio\bio_process.py:12:1: C901 'bio_process' is too complex (16)
neurokit2\complexity\complexity_delay.py:6:1: F401 'pandas as pd' imported but unused
neurokit2\complexity\complexity_delay.py:21:128: E501 line too long (132 > 127 characters)
neurokit2\complexity\complexity_delay.py:23:128: E501 line too long (648 > 127 characters)
neurokit2\complexity\complexity_delay.py:67:128: E501 line too long (306 > 127 characters)
neurokit2\complexity\complexity_delay.py:68:128: E501 line too long (161 > 127 characters)
neurokit2\complexity\complexity_delay.py:69:128: E501 line too long (186 > 127 characters)
neurokit2\complexity\complexity_delay.py:202:9: F841 local variable 'line' is assigned to but never used
neurokit2\complexity\complexity_dimension.py:18:128: E501 line too long (282 > 127 characters)
neurokit2\complexity\complexity_dimension.py:58:128: E501 line too long (160 > 127 characters)
neurokit2\complexity\complexity_dimension.py:230:1: C901 '_embedding_dimension_neighbors' is too complex (12)
neurokit2\complexity\complexity_embedding.py:4:1: F401 'mpl_toolkits.mplot3d' imported but unused
neurokit2\complexity\complexity_embedding.py:12:128: E501 line too long (224 > 127 characters)
neurokit2\complexity\complexity_embedding.py:14:128: E501 line too long (444 > 127 characters)
neurokit2\complexity\complexity_embedding.py:16:128: E501 line too long (532 > 127 characters)
neurokit2\complexity\complexity_embedding.py:26:128: E501 line too long (279 > 127 characters)
neurokit2\complexity\complexity_embedding.py:28:128: E501 line too long (294 > 127 characters)
neurokit2\complexity\complexity_embedding.py:62:128: E501 line too long (306 > 127 characters)
neurokit2\complexity\complexity_optimize.py:11:1: F401 '.complexity_r._optimize_r' imported but unused
neurokit2\complexity\complexity_optimize.py:63:128: E501 line too long (306 > 127 characters)
neurokit2\complexity\complexity_optimize.py:64:128: E501 line too long (161 > 127 characters)
neurokit2\complexity\complexity_optimize.py:65:128: E501 line too long (186 > 127 characters)
neurokit2\complexity\complexity_optimize.py:66:128: E501 line too long (160 > 127 characters)
neurokit2\complexity\complexity_optimize.py:67:128: E501 line too long (210 > 127 characters)
neurokit2\complexity\complexity_optimize.py:293:128: E501 line too long (306 > 127 characters)
neurokit2\complexity\complexity_optimize.py:439:128: E501 line too long (138 > 127 characters)
neurokit2\complexity\complexity_r.py:17:128: E501 line too long (279 > 127 characters)
neurokit2\complexity\complexity_r.py:19:128: E501 line too long (294 > 127 characters)
neurokit2\complexity\complexity_r.py:21:128: E501 line too long (141 > 127 characters)
neurokit2\complexity\complexity_r.py:45:128: E501 line too long (210 > 127 characters)
neurokit2\complexity\complexity_simulate.py:20:128: E501 line too long (131 > 127 characters)
neurokit2\complexity\complexity_simulate.py:162:9: W503 line break before binary operator
neurokit2\complexity\complexity_simulate.py:163:9: W503 line break before binary operator
neurokit2\complexity\entropy_approximate.py:3:1: F401 'pandas as pd' imported but unused
neurokit2\complexity\entropy_approximate.py:12:128: E501 line too long (515 > 127 characters)
neurokit2\complexity\entropy_approximate.py:14:128: E501 line too long (148 > 127 characters)
neurokit2\complexity\entropy_approximate.py:22:128: E501 line too long (279 > 127 characters)
neurokit2\complexity\entropy_approximate.py:24:128: E501 line too long (294 > 127 characters)
neurokit2\complexity\entropy_approximate.py:26:128: E501 line too long (220 > 127 characters)
neurokit2\complexity\entropy_approximate.py:54:128: E501 line too long (210 > 127 characters)
neurokit2\complexity\entropy_approximate.py:55:128: E501 line too long (170 > 127 characters)
neurokit2\complexity\entropy_fuzzy.py:2:1: F401 'numpy as np' imported but unused
neurokit2\complexity\entropy_fuzzy.py:3:1: F401 'pandas as pd' imported but unused
neurokit2\complexity\entropy_fuzzy.py:21:128: E501 line too long (279 > 127 characters)
neurokit2\complexity\entropy_fuzzy.py:23:128: E501 line too long (294 > 127 characters)
neurokit2\complexity\entropy_fuzzy.py:25:128: E501 line too long (182 > 127 characters)
neurokit2\complexity\entropy_multiscale.py:4:1: F401 'pandas as pd' imported but unused
neurokit2\complexity\entropy_multiscale.py:16:128: E501 line too long (209 > 127 characters)
neurokit2\complexity\entropy_multiscale.py:18:128: E501 line too long (245 > 127 characters)
neurokit2\complexity\entropy_multiscale.py:25:128: E501 line too long (370 > 127 characters)
neurokit2\complexity\entropy_multiscale.py:27:128: E501 line too long (294 > 127 characters)
neurokit2\complexity\entropy_multiscale.py:29:128: E501 line too long (182 > 127 characters)
neurokit2\complexity\entropy_multiscale.py:44:128: E501 line too long (213 > 127 characters)
neurokit2\complexity\entropy_multiscale.py:74:128: E501 line too long (249 > 127 characters)
neurokit2\complexity\entropy_sample.py:2:1: F401 'numpy as np' imported but unused
neurokit2\complexity\entropy_sample.py:3:1: F401 'pandas as pd' imported but unused
[... cut ...]
neurokit2\stats\mutual_information.py:169:9: W503 line break before binary operator
neurokit2\stats\mutual_information.py:170:9: W503 line break before binary operator
neurokit2\stats\mutual_information.py:171:9: W503 line break before binary operator
neurokit2\stats\rescale.py:3:1: F401 'pandas as pd' imported but unused
neurokit2\stats\standardize.py:3:1: F401 'pandas as pd' imported but unused
neurokit2\stats\summary.py:4:1: F401 'pandas as pd' imported but unused
neurokit2\stats\summary.py:5:1: F401 'scipy.stats' imported but unused
```

DominiqueMakowski, 2020-06-01:
Unofrunately, neither black nor docformatter wraps the lines of the arguments of the docstring it seems 😕

DominiqueMakowski, 2020-06-01:
@JanCBrammer don't you have an intern or a student interested in learning Python and contributing by doing easy stuff (albeit QUITE tedious) 😁 😅

JanCBrammer, 2020-06-02:
> don't you have an intern or a student interested in learning Python and contributing by doing easy stuff

I wish I had... :D

DominiqueMakowski, 2020-06-13:
I think time has come to try to trigger codecov from the GH workflow rather than from travis, moreover, I wonder if we could include the doctests (i.e., the 'tests' from the examples) into the coverage @Tam-Pham

Tam-Pham, 2020-06-13:
> I think time has come to try to trigger codecov from the GH workflow rather than from travis

Sure. I'm all for migrating the work from travis to GH workflow. So that at one point travis can be removed 😆 I will add the codecov action to the test workflow, after running tox

> I wonder if we could include the doctests (i.e., the 'tests' from the examples) into the coverage 

hmm this I have never heard of. But let's look into it. It will be good if we can

closed 2020-07-09 by DominiqueMakowski